libopencm3/libopencm3-examples #76: reuse steps "don't work"

karlp, 2015-01-19:
To be confirmed, but clearly needs rewording at least..

```
19:40 < Roklobsta> the instructions on https://github.com/libopencm3/libopencm3-examples README 
                   don't work.  "make: *** No rule to make target `miniblink.elf', needed by 
                   `elf'.  Stop."
19:54 < Roklobsta> ah i did mean only the part from REUSE onwards.
19:54 < Roklobsta> the parts before were fine.
19:54 < Roklobsta> sorry should have been clearer.

```

tkln, 2015-03-29:
I came across this behavior and it seems to be triggered by having LDSCRIPT (or OPENCM3_DIR) set to an invalid value in the project Makefile. The reuse instructions aren't exactly straightforward and the disjoint between the apparent cause and effect of this make it a relatively nasty thing to run into.

ChuckM, 2015-03-29:
This has been true forever, but it would be nice to fix it. When ever any dependency on the final link stage can't be me  (either the directory is mis-pointed or the library hasn't been built) the error message is very non-intutitive.

mox-mox, 2015-05-01:
So, for those trying the reuse steps for the first time, what this means is, that you will have to do a "make" in the libopencm3 subfolder you downloaded in step 2. Then, everything should work fine.

plaes, 2015-06-28:
I have patch for that in pull #89

karlp, 2015-06-29:
This ticket still needs to be open to clarify any wording to requiring a make in the lib dir first right?

karlp, 2018-08-29:
This all "works" per the readme now, but it's still more complicated than necessary.  Perhaps a better template repo would be more useful.

closed 2018-08-29 by karlp