libopencm3/libopencm3-examples #86: Alignment for USB structs

rkreis, 2015-05-01:
Various USB examples contain structs like `usb_device_descriptor` which are passed to `usb_copy_to_pm`, which will read them 16 bit at a time. If the struct happens not to be aligned, this causes a performance penalty on Cortex-M3 upwards, and even breaks on Cortex-M0 which doesn't support unaligned access at all (I ran into this while porting an USB example to STM32L0). gcc supports `__attribute__ ((aligned (2)))` for variables, which could be used to ensure proper alignment. I don't know if using this feature is worth it for the existing examples, but we definitely have to keep this in mind when USB examples are ported to controllers with an M0 core.

karlp, 2015-10-03:
So wish I'd remembered this ticket you filed earlier :)

karlp, 2016-10-01:
We've merged code that handles this, by doing bytewise copies whenever necessary. This leaves room for potential perf optimizations later, but correct and robust should be first.

closed 2016-10-01 by karlp