libopencm3/libopencm3-examples #87: Communicating arm-none-eabi-gcc location on windows

andrey-gvrd, 2015-05-17:
make in top level directory fails with this error:

    make[2]: arm-none-eabi-gcc: Command not found

arm-none-eabi-gcc.exe in my system is located at:

    C:\Program Files (x86)\GNU Tools ARM Embedded\4.9 2015q1\bin

I've tried running this command:

    export PATH="/c//Python27:/c//Program Files (x86)/GNU Tools ARM Embedded/4.9 2015q1/bin:/usr/local/bin:/usr/bin:/bin"

But it doesn't help. What am I doing wrong?

ChuckM, 2015-05-17:
If you type simply `arm-none-eabi-gcc --version` on the command line what does it say?

andrey-gvrd, 2015-05-17:
...

    arm-none-eabi-gcc: command not found

andrey-gvrd, 2015-05-17:
I've reinstalled it to a folder which contains no space characters and that part now works.

andrey-gvrd, 2015-05-17:
I believe I'm still stuck on the same issue. 

Even though `arm-none-eabi-gcc --version` now returns `arm-none-eabi-gcc.exe (GNU Tools for ARM Embedded Processors) 4.9.3 20150303 (release) [ARM/embedded-4_9-branch revision 221220]`, make now fails with `flash.d:1: *** multiple target patterns.  Stop.`

I've tried setting paths this way:

     export PATH="/cygwin/c/Python27:/cygwin/c/ARMToolchain/bin:/usr/local/bin:/usr/bin:/bin"

But it didn't work.

ChuckM, 2015-05-17:
Can you pop over to the #libopencm3 channel on irc.freenode.net ?

karlp, 2015-08-11:
Is there anything real here to work on? Was this resolved?

andrey-gvrd, 2015-08-13:
I think I settled on the fact that I have to use linux to make this work. So yeah, issue's done.

closed 2015-08-13 by andrey-gvrd