libopencm3/libopencm3-examples #99: stlink-flash target not supported?

tdicola, 2015-08-04:
I was looking at reusing the makefiles from the examples here and noticed the stlink-flash target doesn't seem to be fully setup.  In Makefile.rules there's a definition for the %.stlink-flash target that runs st-flash here: https://github.com/libopencm3/libopencm3-examples/blob/master/examples/Makefile.rules#L197-L199  However I don't see anything in the file that depends on a $(BINARY).stlink-flash target which would trigger that flashing rule.  

I think there should be something like `stlink-flash: $(BINARY).stlink-flash` added to this block that generates these flashing targets: https://github.com/libopencm3/libopencm3-examples/blob/master/examples/Makefile.rules#L134-L135  That way you could run 'make stlink-flash' in an example folder and it would use st-flash to flask the binary.  Just wanted to check am I missing something else that adds that rule, or would this be a useful pull to send back to this repo?

Thanks for providing the great library & examples BTW!

karlp, 2015-08-11:
I'm actually much more inclined to just remove all of the flash helping snippets from the makefiles.  They have various attempts to support both texane/stlink and openocd and black magic probe, and they try and support various debug hardware for openocd as well.  I plan on just updating the readme with a few links on how to flash, but it's kinda out of scope, and always relied on out of tree software to be installed  correctly.

karlp, 2015-08-11:
Closing as a duplicate of #34

closed 2015-08-11 by karlp

tdicola, 2015-08-13:
Makes sense, thanks!