libopencm3/libopencm3-examples #107: usb-cdc for maple-mini not compilng

advent-embedded, 2015-10-26:
when i tried compiling the cdc-acm project i get an error stating "stm32f103_usb_driver undeclared " in line no. 261 (cdcacm.c source file) the usbd_init function, as i tried declaring the variable of type usbd_driver which is typedef structure in usb_private.h still get error as storage size unknown.

kuldeepdhaka, 2015-10-26:
Are you sure that the you have the right library compiled against example?
also, looking at the makefile's, it appear that it should not appear.
and, trying to declare "stm32f103_usb_driver" is useless because it will never contain the backend functions.
conclusion: please provide more details.

advent-embedded, 2015-11-02:
Thanks for your reply, and yes i do have compiled the libopencm3 project successfully. please view my compile log 

cdcacm.c: In function 'cdcacm_set_config':
cdcacm.c:234:5: warning: passing argument 4 of 'usbd_register_control_callback' from incompatible pointer type
     cdcacm_control_request);
     ^
In file included from cdcacm.c:23:0:
../../../../../libopencm3//include/libopencm3/usb/usbd.h:94:12: note: expected 'usbd_control_callback' but argument is of type 'int (*)(struct usbd_device *, struct usb_setup_data *, uint8_t **, uint16_t *, void (**)(struct usbd_device *, struct usb_setup_data *))'
 extern int usbd_register_control_callback(usbd_device *usbd_dev, uint8_t type,
            ^
cdcacm.c: In function 'main':
cdcacm.c:261:24: error: 'stm32f103_usb_driver' undeclared (first use in this function)
  usbd_dev = usbd_init(&stm32f103_usb_driver, &dev, &config, usb_strings, 3, usbd_control_buffer, sizeof(usbd_control_buffer));
                        ^
cdcacm.c:261:24: note: each undeclared identifier is reported only once for each function it appears in
make: *** [cdcacm.o] Error 1

ChuckM, 2015-11-02:
Can you check what commit level your libopencm3 sub-module is in your examples directory? Sometimes as changes are happening there are periods where the examples are out of date with the current level of the library.

advent-embedded, 2015-11-03:
yep my project is with latest commit and no problem with that, problem was only with the source i hope, running through the source code of cdcacm.c, i didn't see 'stm32f103_usb_driver' variable (pointer argument for usbd_init function was not declared anywhere in the source.

advent-embedded, 2015-11-03:
Compile Log
cdcacm.c: In function 'main':
cdcacm.c:260:24: error: 'stm32f103_usb_driver' undeclared (first use in this function)
  usbd_dev = usbd_init(&stm32f103_usb_driver, &dev, &config, usb_strings, 3, usbd_control_buffer, sizeof(usbd_control_buffer));
                        ^
cdcacm.c:260:24: note: each undeclared identifier is reported only once for each function it appears in
make: *** [cdcacm.o] Error 1

advent-embedded, 2015-11-03:
please some one give me solution to resolve this error i ve been stuck for past 5 days on this.
i tried declaring the variable, then i get as Storage size unknown for the variable, and then i tried including usb-private.h still getting some gig errors

karlp, 2015-11-03:
@advent-embedded no-one else can reproduce it.  Your "logs" are far too short to provide any context.  If you run "git status" in /your/source/libopencm3-examples does it say you have changes in the submodule or in the project?  If you run `git diff origin/master` do you have any changes?  Do you run "make clean; make" from the root directory?  Are you doing anything else that might seem obvious to you but not to us?  Here's a clean checkout showing it working.  The sort of error you're reporting shouldn't be caused by any toolchain problems, so it still very much sounds like you have an unclean environment.

```
karlp@cuth:~/src$ mkdir test-advent
karlp@cuth:~/src$ cd test-advent/
karlp@cuth:~/src/test-advent$ git clone git@github.com:libopencm3/libopencm3-examples.git
Cloning into 'libopencm3-examples'...
Host key fingerprint is SHA256:nThbg6kXUpJWGl7E1IGOCspRomTxdCARLviKw6E5SY8
+---[RSA 2048]----+
| =+o...+=o..     |
|o++... *o .      |
|*.o.  *o.        |
|oo.  ..o.= .     |
|.+o. .. S =      |
|*=+ .  o = .     |
|OE .  . o        |
| o     .         |
|                 |
+----[SHA256]-----+

remote: Counting objects: 3319, done.
remote: Total 3319 (delta 0), reused 0 (delta 0), pack-reused 3319
Receiving objects: 100% (3319/3319), 737.41 KiB | 258.00 KiB/s, done.
Resolving deltas: 100% (1891/1891), done.
Checking connectivity... done.
karlp@cuth:~/src/test-advent$ cd libopencm3-examples/
karlp@cuth:~/src/test-advent/libopencm3-examples (master)$ make
######## ERROR ########
	libopencm3 is not initialized.
	Please run:
	$ git submodule init
	$ git submodule update
	before running make.
######## ERROR ########
Makefile:56: recipe for target 'lib' failed
make: *** [lib] Error 1
karlp@cuth:~/src/test-advent/libopencm3-examples (master)$ git submodule init
Submodule 'libopencm3' (https://github.com/libopencm3/libopencm3.git) registered for path 'libopencm3'
karlp@cuth:~/src/test-advent/libopencm3-examples (master)$ git submodule update
Cloning into 'libopencm3'...
remote: Counting objects: 18901, done.
remote: Compressing objects: 100% (14/14), done.
remote: Total 18901 (delta 5), reused 0 (delta 0), pack-reused 18887
Receiving objects: 100% (18901/18901), 5.39 MiB | 343.00 KiB/s, done.
Resolving deltas: 100% (10566/10566), done.
Checking connectivity... done.
Submodule path 'libopencm3': checked out '43e66927fc48ffb9ad90fec275290ea7c0f88781'
karlp@cuth:~/src/test-advent/libopencm3-examples (master)$ make
  GENHDR  include/libopencm3/stm32/f4/irq.json
  GENHDR  include/libopencm3/stm32/f1/irq.json
  GENHDR  include/libopencm3/stm32/f3/irq.json
  GENHDR  include/libopencm3/stm32/f2/irq.json
  GENHDR  include/libopencm3/stm32/f0/irq.json
  GENHDR  include/libopencm3/stm32/l1/irq.json
  GENHDR  include/libopencm3/stm32/l0/irq.json
.... finishes cleanly here after a while....
```

kuldeepdhaka, 2015-11-03:
@advent-embedded try this
https://github.com/libopencm3/libopencm3/blob/master/lib/stm32/st_usbfs_v1.c#L30
this is the actual driver that will be used.
include via `#include <libopencm3/stm32/st_usbfs.h>` and replace `stm32f103_usb_driver` with `st_usbfs_v1_usb_driver` and it should compile then.

karlp, 2015-11-03:
@advent-embedded for some background, the current HEAD of master in the -examples repo is _NOT_ uptodate with the current HEAD of master of the libopencm3 repo.  This will be brought up todate "real soon now" but the -examples repo itself, _is_ correct with the submodule it references.  You should only be having problems if you're trying to mix "newer" versions of the library with "older" versions of the examples.

advent-embedded, 2015-11-05:
@karlp Thanks for your reply, i cloned the "libopnecm3-examples" repository using the https URL given there, and later by getting inside the "libopencm3-examples" directory i cloned the "libopnencm3" project thus getting the same directory structure, (libopencm3 project inside the libopencm3-examples directory) and run make inside the libopencm3 library directory which was successful, and when i tried compiling the examples project i got stuck with that issue.

Whats the real pitfall in my procedure.

advent-embedded, 2015-11-05:
Here is my git repo diff status

kannan@samsung:~/Documents/libopencm3-examples$ git status
# On branch master
# Changes not staged for commit:
#   (use "git add <file>..." to update what will be committed)
#   (use "git checkout -- <file>..." to discard changes in working directory)
#
#	modified:   libopencm3 (new commits)
#
no changes added to commit (use "git add" and/or "git commit -a")
kannan@samsung:~/Documents/libopencm3-examples$ git diff master
diff --git a/libopencm3 b/libopencm3
index 43e6692..e6f267e 160000
--- a/libopencm3
+++ b/libopencm3
@@ -1 +1 @@
-Subproject commit 43e66927fc48ffb9ad90fec275290ea7c0f88781
+Subproject commit e6f267e10363010f9607cf5067e30ddf8f964530
kannan@samsung:~/Documents/libopencm3-examples$

karlp, 2015-11-05:
Seee that bit where it says, "modified: libopencm3 (new commits)" that means you have a mismatched library, most likely a newer one.  This is _exactly_ what was suggested initially.  There are _ongoing_ api changes in the current HEAD of the library, and the -examples repo has _not_ yet been getting updated at the same time.

karlp, 2015-11-05:
The real pitfall was you shoudl have just typed "make" in the root directory of the -examples repo and followed itðs instructions, straight after cloning.  You are totally welcome to  mismatch versions, it's exactly how most development is done, but it's not very interesting to file bugs for things not working when you do so.

closed 2015-11-05 by karlp