libopencm3/libopencm3-examples #111: terse

J-Dunn, 2015-12-22:
I've just started diving into opencm3 and I'm pretty impressed. I quickly got a few demos built and flashed. I started with the stm32f429i-discovery which has quite a lot of good stuff. 

However, I quickly got stuck because the readme's are so terse.

The top level readme has: 
>>
2. usart\_console - Program a USART on the board as a console (requires
   clonsing jumpers on your discovery board connecting USART1 to the
   programmer)
>>

Well I guess clonsing is closing but they are already in the closed position to use the STLINK to flash the board. Did this mean open? But that does not work either. :(

The full readme for that demo says: 
>>
Simple USART Example
--------------------

This example sets up a USART port and provides a few simple
character handling functions to that a short interactive program
can be demonstrated. It re-uses the clock setup from systick_blink
and that means you could do times delays but that aspect isn't used.
>>

So what should I expect / do  ? 

If I get nothing is there a problem? I would not know because there is so little about what this demo is supposed to do. 

In fact this whole jumper story is completely opaque, even in the ST user manual . On or off I never get a ACM device showing. There seems to be more to it but some explicit instructions are needed to get the demo to run. 

Some of the readme's in the other branches are pretty good but this branch is a little too light to of much help.

ChuckM, 2015-12-22:
The F429 disco board does not enumerate the serial port through the debug USB port, you need to attach a TTL/Serial adapter to the pins in question (something like the AdaFruit FTDI Friend (https://www.adafruit.com/products/284)). The Nucleo boards and the F469 Disco board have the STLink V2+ firmware (which is mbed compatible) and provides the serial port through the debug USB connection.

J-Dunn, 2015-12-22:
Thanks for the reply Chuck , clearly this needs to be in the readme. 

How should I access the serial port via STlink? I'm only seeing one device on 429i. I think F411RE provides a MSD as well as seperate device. 

I also have a MAX3232 I can hook up but a bit more info is required to get anywhere here. 

It's good stuff, just needs more explanation to be useful.

ChuckM, 2015-12-22:
Looking at the ST site it looks like they have created an upgraded ST Link package for the 429Disco I'll have to wait until I get back to where my 429 board is to upgrade it and see how it changes what the examples in LOC3 should do.

J-Dunn, 2015-12-24:
There seems to some erroneous suggestion that the jumpers need changing. 

The datasheets for this series of boards says that the purpose of the jumpers is whether the STLINK debugs the board to which it is attached or is available as a stand alone JTAG debugger on the side connector. In which case it can be used to debug an independent board and is separate from the Discovery board.

It seems that this needs correcting in several files. 

Regards.

ChuckM, 2015-12-25:
So I did a bit of research on this, and there are two revs of the hardware, the original (Rev A.) and the latest (Rev B.). The Rev B. board already has the serial port connected to the STLink chip and has the STLink V2.1-B software on it. The Rev A. board does *not* have the SB15 and SB11 bridges "fitted" (at least according to the User Guide) and there is a 2-pin header  JP4 which  connects to the TX and RX pins of the STLink processor (note putting a jumper on it shorts TX to RX so don't do that). Both schematics show PA9 and PA10 as the pins used. 

If you flash the upgraded STLink software you should see the three devices appear on your computer when it is connected to the host/debug port. Its on my list of things to do as soon I can get access to that board again. 

Since JP4 is **after** the solder bridges, you can see if they have been closed by measuring the continuity between the JP4 pin, and the PA9 (or PA10) pin on the discovery board connectors.

J-Dunn, 2015-12-25:
Thanks for digging into this. FWIW, I have a M1075B board . It has three, two-pin jumpers JP1..3 , it also has a pair of two, two-pin jumpers ( physically the same thing ) which are labelled CN4, JP4 :?

All these have a connectors closing the link.

There are also two vias without pins labelled JP4. They have solderable bridges close by and are labelled TX, RX. It seems that matches what you describe. 

It produces just one new device with one configuration 
```
[100196.713034] usb 2-5: new full-speed USB device number 52 using ohci-pci
[100196.916062] usb 2-5: New USB device found, idVendor=0483, idProduct=3748
[100196.916076] usb 2-5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[100196.916083] usb 2-5: Product: STM32 STLink

lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 052: ID 0483:3748 STMicroelectronics ST-LINK/V2
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
```
It was purchased a while ago , I guess it has the earlier STLINK firmware.

It seems any comments in readme need to be accompanied with a warning about possible variations of hardware and software with some RTFM , YMMV caveats ;)

ChuckM, 2015-12-25:
In that case you just need the new firmware, http://www.st.com/web/en/catalog/tools/PF260217

J-Dunn, 2015-12-25:
Ah, thanks. Now I know why my F411RE board has a different  USB product ID . 

It has v2.1 STLINK. I thought it was because it was a different product. 
```
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 005: ID 0483:374b STMicroelectronics ST-LINK/V2.1 (Nucleo-F103RB)
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
```

It is the STLINK version , not the board : 374b is v2.1 ; 3748 is v2

ChuckM, 2015-12-25:
Exactly right. You can think of all these boards as two different boards joined at the Software Debug lines, an STLink board and the target board. 

The STLink software is not open source, however you can also install the BlackMagic software on the STLink processor (see https://github.com/blacksphere/blackmagic) you may need to do a small bit of porting to the STLink design on your board, and it represents to the host OS as two serial ports (one debug port and one running the gdb server) but on the plus side you don't need OpenOCD if you're using it.

J-Dunn, 2015-12-26:
Thanks, I'd seen the Black Magic probe but did not see much reason to buy a $60 probe to talk to a $20 board which comes with built-in debugger. One could even buy a Nucleo board for about $12 and snap off the STLINK section to get a cheap, compact debugger probe.

However, if I can flash the STLINK with OSS code, that's an interesting possibility. It also sounds like a good way to brick the board if I don't get the porting just right ;)

ChuckM, 2015-12-26:
Actually the given the way the debug port works you can't brick it, you can always use another board to reprogram the factory firmware.

karlp, 2017-02-27:
Unless there are further issues to be resolved here, I'd like to close this ticket. Please reopen if you feel there's more that needs to be done.

closed 2017-02-27 by karlp