libopencm3/libopencm3-examples #112: stm32f429i-discovery/sdram not working

J-Dunn, 2015-12-22:
I built and flashed the stm32f429i-discovery/sdram demo. 

I get a mass storage device showing up and it auto-mounts as vfat on Fedora 23.  But if I try to do anything like cat or head on the datafile the I see on the drive I get an IO error.

ChuckM, 2015-12-22:
This is a bit confusing, are you talking about the STMicro demo or are you talking about the libopencm3 demo? In the latter case the demo sets up a very simple memory tool for looking at the memory over the serial port, it doesn't do anything with the USB port.

J-Dunn, 2015-12-22:
Sorry I meant ramdisk not sdram .  Lack of a readme in the latter may have lead to difficulty in identifying it. 

If I flash the device I get a new mass storage device:

ls /dev/sdc*
/dev/sdc

this gets auto-mounted at /run/media/user/RAMDISK: 

but if I so much as try to ls anything on it, it falls apart:

# ls /run/media/user/RAMDISK/*
# dmesg
[127168.691075] sd 12:0:0:0: Attached scsi generic sg3 type 0
[127168.700118] sd 12:0:0:0: [sdc] 1024 512-byte logical blocks: (524 kB/512 KiB)
[127168.707150] sd 12:0:0:0: [sdc] Write Protect is off
[127168.707169] sd 12:0:0:0: [sdc] Mode Sense: 03 00 00 ff
[127168.714120] sd 12:0:0:0: [sdc] No Caching mode page found
[127168.714139] sd 12:0:0:0: [sdc] Assuming drive cache: write through
[127168.764069]  sdc:
[127168.801055] sd 12:0:0:0: [sdc] Attached SCSI removable disk
[127168.801055] sd 12:0:0:0: [sdc] Attached SCSI removable disk
[127199.880036] usb 2-4: reset full-speed USB device number 92 using ohci-pci
[127230.910050] usb 2-4: reset full-speed USB device number 92 using ohci-pci
[127241.253050] usb 2-4: reset full-speed USB device number 92 using ohci-pci
[127257.601035] usb 2-4: reset full-speed USB device number 92 using ohci-pci
[127257.945050] usb 2-4: reset full-speed USB device number 92 using ohci-pci
[127268.285053] usb 2-4: reset full-speed USB device number 92 using ohci-pci
[127268.468522] sd 12:0:0:0: Device offlined - not ready after error recovery
[127268.468555] sd 12:0:0:0: [sdc] FAILED Result: hostbyte=DID_ABORT driverbyte=DRIVER_OK
[127268.468565] sd 12:0:0:0: [sdc] CDB: Write(10) 2a 00 00 00 00 00 00 00 01 00
[127268.468572] blk_update_request: I/O error, dev sdc, sector 0
[127268.468579] Buffer I/O error on dev sdc, logical block 0, lost sync page write
[127268.468623] sd 12:0:0:0: rejecting I/O to offline device
[127268.468631] sd 12:0:0:0: killing request
[127268.566261] sd 12:0:0:0: rejecting I/O to offline device
[127268.566306] sd 12:0:0:0: rejecting I/O to offline device
[127268.566317] FAT-fs (sdc): FAT read failed (blocknr 1)

ChuckM, 2015-12-23:
Ah, I see. So there are a couple of things here which could use some clarity. First, there are **two** ARM chips on the Discovery boards, one has the STLink software and the other is the target device. 

Next, the latest STLINK (V2.1-1) software is mBed compatible. That means when it plugs in you should see three devices. On my system with the 32F469 discovery board I see this:
```
[350671.782834] usb 1-1.2: new full-speed USB device number 6 using ehci-pci
[350671.878098] usb 1-1.2: New USB device found, idVendor=0483, idProduct=374b
[350671.878103] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[350671.878106] usb 1-1.2: Product: STM32 STLink
[350671.878109] usb 1-1.2: Manufacturer: STMicroelectronics
[350671.878111] usb 1-1.2: SerialNumber: 0670FF544949847067092343
[350671.931978] usb-storage 1-1.2:1.1: USB Mass Storage device detected
[350671.932242] scsi5 : usb-storage 1-1.2:1.1
[350671.999513] cdc_acm 1-1.2:1.2: This device cannot do calls on its own. It is not a modem.
[350671.999544] cdc_acm 1-1.2:1.2: ttyACM0: USB ACM device
[350672.000082] usbcore: registered new interface driver cdc_acm
[350672.000084] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
[350672.932657] scsi 5:0:0:0: Direct-Access     MBED     microcontroller  1.0  PQ: 0 ANSI: 2
[350672.933121] sd 5:0:0:0: Attached scsi generic sg4 type 0
[350672.933939] sd 5:0:0:0: [sdd] 4128 512-byte logical blocks: (2.11 MB/2.01 MiB)
[350672.935096] sd 5:0:0:0: [sdd] Write Protect is off
[350672.935102] sd 5:0:0:0: [sdd] Mode Sense: 03 00 00 00
[350672.936508] sd 5:0:0:0: [sdd] No Caching mode page found
[350672.936513] sd 5:0:0:0: [sdd] Assuming drive cache: write through
[350672.941342] sd 5:0:0:0: [sdd] No Caching mode page found
[350672.941348] sd 5:0:0:0: [sdd] Assuming drive cache: write through
[350672.946946]  sdd:
[350672.950020] sd 5:0:0:0: [sdd] No Caching mode page found
[350672.950023] sd 5:0:0:0: [sdd] Assuming drive cache: write through
[350672.950026] sd 5:0:0:0: [sdd] Attached SCSI removable disk
```

The target device can be programmed by dropping a binary image on to the mass storage device. The device has to be 'nocache', 'noatime' mounted. You can also connect to it using OpenOCD which will open up a gdb server on port 3333, so you can connect using localhost:3333 as the remote target. And finally the /dev/ttyACMx (usually 0 if its the only one) serial port which is connected to one of the USARTs on the target chip, typically on the Port A pins. 

Since the '429 board was not originally designed to run mbed compatible firmware you may have to make jumper changes for it to work correctly. There are also updated windows USB drivers (if you have a windows client) which recognize the diagnostic (STLINK) port, the serial port, and the disk drive.

In general I ignore the mass storage device, program it using gdb through connection with OpenOCD and use gnu screen to connect to the serial port with `screen /dev/ttyACM0 115200`.

J-Dunn, 2015-12-23:
Thanks. I'm working on Linux too using openOCD and gdb. We are on the same page with that. 

I have flashed several of the examples from this project , so the default software devices are not relevant beyond what is provided by the STLINK chip.

I currently have the ramdisk demo on the board.  If I plug in just the STLINK, I get one device. It's a V2  but I don't know a means of getting the minor version numbers.

```
[168005.809035] usb 2-6: new full-speed USB device number 113 using ohci-pci
[168006.012061] usb 2-6: New USB device found, idVendor=0483, idProduct=3748
[168006.012073] usb 2-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[168006.012079] usb 2-6: Product: STM32 STLink
[168006.012085] usb 2-6: Manufacturer: STMicroelectronics
[168006.012090] usb 2-6: SerialNumber: xxxxxxxxxxxxxx
bash-4.3#lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 113: ID 0483:3748 STMicroelectronics ST-LINK/V2
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
```
Now , if I plug in the OTG: (note the for clarity I'm using a different USB port ). It is correctly detected as mass storage, no cache device. Then it starts to go wroing:
```
[168326.393034] usb 2-4: new full-speed USB device number 114 using ohci-pci
[168326.585889] usb 2-4: New USB device found, idVendor=0483, idProduct=5741
[168326.585903] usb 2-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[168326.585910] usb 2-4: Product: MSC Demo
[168326.585915] usb 2-4: Manufacturer: Black Sphere Technologies
[168326.585920] usb 2-4: SerialNumber: DEMO
[168326.591083] usb-storage 2-4:1.0: USB Mass Storage device detected
[168326.597329] scsi host23: usb-storage 2-4:1.0
[168327.605922] scsi 23:0:0:0: Direct-Access     VendorID ProductID        0.00 PQ: 0 ANSI: 4
[168327.609657] sd 23:0:0:0: Attached scsi generic sg3 type 0
[168327.619940] sd 23:0:0:0: [sdc] 1024 512-byte logical blocks: (524 kB/512 KiB)
[168327.626930] sd 23:0:0:0: [sdc] Write Protect is off
[168327.626946] sd 23:0:0:0: [sdc] Mode Sense: 03 00 00 ff
[168327.633926] sd 23:0:0:0: [sdc] No Caching mode page found
[168327.633942] sd 23:0:0:0: [sdc] Assuming drive cache: write through
[168327.684895]  sdc:
[168327.721871] sd 23:0:0:0: [sdc] Attached SCSI removable disk
[168358.919219] usb 2-4: reset full-speed USB device number 114 using ohci-pci
[168389.886044] usb 2-4: reset full-speed USB device number 114 using ohci-pci
[168400.227040] usb 2-4: reset full-speed USB device number 114 using ohci-pci
[168416.575052] usb 2-4: reset full-speed USB device number 114 using ohci-pci
[168416.917053] usb 2-4: reset full-speed USB device number 114 using ohci-pci
[168427.257046] usb 2-4: reset full-speed USB device number 114 using ohci-pci
[168427.440297] sd 23:0:0:0: Device offlined - not ready after error recovery
[168427.440331] sd 23:0:0:0: [sdc] FAILED Result: hostbyte=DID_ABORT driverbyte=DRIVER_OK
[168427.440341] sd 23:0:0:0: [sdc] CDB: Write(10) 2a 00 00 00 00 00 00 00 01 00
[168427.440348] blk_update_request: I/O error, dev sdc, sector 0
[168427.440355] Buffer I/O error on dev sdc, logical block 0, lost sync page write
[168427.440394] sd 23:0:0:0: rejecting I/O to offline device
[168427.440403] sd 23:0:0:0: killing request
[168427.567226] sd 23:0:0:0: rejecting I/O to offline device
[168427.567307] sd 23:0:0:0: rejecting I/O to offline device
[168427.567319] FAT-fs (sdc): FAT read failed (blocknr 1)
```
/var/log/messages contains some more information:
```
Dec 23 06:57:09 localhost kernel: sd 23:0:0:0: Attached scsi generic sg3 type 0
Dec 23 06:57:09 localhost kernel: sd 23:0:0:0: [sdc] 1024 512-byte logical blocks: (524 kB/512 KiB)
Dec 23 06:57:09 localhost kernel: sd 23:0:0:0: [sdc] Write Protect is off
Dec 23 06:57:09 localhost kernel: sd 23:0:0:0: [sdc] No Caching mode page found
Dec 23 06:57:09 localhost kernel: sd 23:0:0:0: [sdc] Assuming drive cache: write through
Dec 23 06:57:09 localhost kernel: sdc:
Dec 23 06:57:09 localhost kernel: sd 23:0:0:0: [sdc] Attached SCSI removable disk
Dec 23 06:57:40 localhost kernel: usb 2-4: reset full-speed USB device number 114 using ohci-pci
^[[A
Dec 23 06:58:11 localhost kernel: usb 2-4: reset full-speed USB device number 114 using ohci-pci
Dec 23 06:58:22 localhost kernel: usb 2-4: reset full-speed USB device number 114 using ohci-pci
Dec 23 06:58:38 localhost kernel: usb 2-4: reset full-speed USB device number 114 using ohci-pci
Dec 23 06:58:38 localhost kernel: usb 2-4: reset full-speed USB device number 114 using ohci-pci
Dec 23 06:58:49 localhost kernel: usb 2-4: reset full-speed USB device number 114 using ohci-pci
Dec 23 06:58:49 localhost kernel: sd 23:0:0:0: Device offlined - not ready after error recovery
Dec 23 06:58:49 localhost kernel: sd 23:0:0:0: [sdc] FAILED Result: hostbyte=DID_ABORT driverbyte=DRIVER_OK
Dec 23 06:58:49 localhost kernel: sd 23:0:0:0: [sdc] CDB: Write(10) 2a 00 00 00 00 00 00 00 01 00
Dec 23 06:58:49 localhost kernel: blk_update_request: I/O error, dev sdc, sector 0
Dec 23 06:58:49 localhost kernel: Buffer I/O error on dev sdc, logical block 0, lost sync page write
Dec 23 06:58:49 localhost kernel: sd 23:0:0:0: rejecting I/O to offline device
Dec 23 06:58:49 localhost kernel: sd 23:0:0:0: killing request
Dec 23 06:58:49 localhost udisksd[1630]: Mounted /dev/sdc at /run/media/user/RAMDISK on behalf of uid 1000
Dec 23 06:58:49 localhost org.gtk.vfs.UDisks2VolumeMonitor: disc.c:350: error opening file BDMV/index.bdmv
Dec 23 06:58:49 localhost kernel: sd 23:0:0:0: rejecting I/O to offline device
Dec 23 06:58:49 localhost kernel: sd 23:0:0:0: rejecting I/O to offline device
Dec 23 06:58:49 localhost kernel: FAT-fs (sdc): FAT read failed (blocknr 1)
Dec 23 06:58:49 localhost org.gtk.vfs.UDisks2VolumeMonitor: disc.c:350: error opening file BDMV/BACKUP/index.bdmv
```

It looks like some kind of udisk daemon is screwing things up while auto-mounting the device or that the device is not responding correctly.

J-Dunn, 2015-12-23:
This may be a gfvs bug :
https://bbs.archlinux.org/viewtopic.php?pid=1383084

J-Dunn, 2015-12-23:
If I umount /dev/sdc and replug it I can get gfvs off my back. But then a manual mount does not get any further. 

```
mount -t vfat /dev/sdc /home/tmp
mount: /dev/sdc is not a valid block device
```

```Dec 23 07:43:33 localhost kernel: sd 28:0:0:0: [sdc] Attached SCSI removable disk
Dec 23 07:44:04 localhost kernel: usb 2-4: reset full-speed USB device number 117 using ohci-pci
Dec 23 07:44:35 localhost kernel: usb 2-4: reset full-speed USB device number 117 using ohci-pci
Dec 23 07:44:46 localhost kernel: usb 2-4: reset full-speed USB device number 117 using ohci-pci
Dec 23 07:45:02 localhost kernel: usb 2-4: reset full-speed USB device number 117 using ohci-pci
Dec 23 07:45:02 localhost kernel: usb 2-4: reset full-speed USB device number 117 using ohci-pci
Dec 23 07:45:13 localhost kernel: usb 2-4: reset full-speed USB device number 117 using ohci-pci
Dec 23 07:45:13 localhost kernel: sd 28:0:0:0: Device offlined - not ready after error recovery
Dec 23 07:45:13 localhost kernel: sd 28:0:0:0: [sdc] FAILED Result: hostbyte=DID_ABORT driverbyte=DRIVER_OK
Dec 23 07:45:13 localhost kernel: sd 28:0:0:0: [sdc] CDB: Write(10) 2a 00 00 00 00 00 00 00 01 00
Dec 23 07:45:13 localhost kernel: blk_update_request: I/O error, dev sdc, sector 0
Dec 23 07:45:13 localhost kernel: Buffer I/O error on dev sdc, logical block 0, lost sync page write
Dec 23 07:45:13 localhost kernel: sd 28:0:0:0: rejecting I/O to offline device
Dec 23 07:45:13 localhost kernel: sd 28:0:0:0: killing request
```

Have you been able to mount the ramdisk device provided in the demo?