libopencm3/libopencm3-examples #133: stm32/f3/stm32f3-discovery/adc/adc.c hang on hardware at adc_power_on(ADC1);

elemhsb, 2016-04-22:
adc_power_on(ADC1) hangs on while (!adc_is_power_on(adc));

I use gcc-arm-none-eabi-4_9-2015q3 toolchain from https://launchpad.net/gcc-arm-embedded/4.9/4.9-2015-q3-update/+download/gcc-arm-none-eabi-4_9-2015q3-20150921-linux.tar.bz2

Which toolchain should i use ?

karlp, 2016-04-22:
Probably related to https://github.com/libopencm3/libopencm3/issues/654

elemhsb, 2016-05-10:
works now, after fixing libopencm3/libopencm3#654
I did not understand why the ADC is working without enable regulator.
Here a patch for the example to run with regulator.

```
index f66d811..2a4b506 100644
--- a/examples/stm32/f3/stm32f3-discovery/adc/adc.c
+++ b/examples/stm32/f3/stm32f3-discovery/adc/adc.c
@@ -52,7 +52,12 @@ static void adc_setup(void)
        //ADC
        gpio_mode_setup(GPIOA, GPIO_MODE_ANALOG, GPIO_PUPD_NONE, GPIO0);
        gpio_mode_setup(GPIOA, GPIO_MODE_ANALOG, GPIO_PUPD_NONE, GPIO1);
+
        adc_power_off(ADC1);
+       adc_enable_regulator(ADC1);
+       /* Wait for regulator starting up. */
+       { int i; for (i = 0; i < 800000; i++) __asm__("nop"); }
+
        adc_set_clk_prescale(ADC1, ADC_CCR_CKMODE_DIV2);
        adc_set_single_conversion_mode(ADC1);
        adc_disable_external_trigger_regular(ADC1);
```

karlp, 2016-10-01:
Please file separate issues (if they still exist) as separate tickets.  The fixes for adc power on/off were merged into the main library, and the examples have been updated to include that.

closed 2016-10-01 by karlp