libopencm3/libopencm3-examples #134: error: unknown type name 'cookie_io_functions_t' when trying to compile on Arch Linux with GCC 5.3.x

SpotlightKid, 2016-05-02:
I'm trying to compile libopencm3-examples on Arch Linux (`LC_ALL=C make V=1`), and I get the following error: when the make process reaches `examples/stm32/f0/stm32f0-discovery/usart_stdio`:

```
make[1]: Entering directory '/home/chris/src/stm32/libopencm3-examples/examples/stm32/f0/stm32f0-discovery/usart_stdio'
Using /home/chris/src/stm32/libopencm3-examples/libopencm3 path to library
arm-none-eabi-gcc -Os -g -Wextra -Wshadow -Wimplicit-function-declaration -Wredundant-decls -Wmissing-prototypes -Wstrict-prototypes -fno-common -ffunction-sections -fdata-sections -MD -Wall -Wundef -I/home/chris/src/stm32/libopencm3-examples/libopencm3/include -DSTM32F0 -mthumb -mcpu=cortex-m0 -msoft-float -o usart_stdio.o -c usart_stdio.c
usart_stdio.c: In function 'usart_setup':
usart_stdio.c:67:2: error: unknown type name 'cookie_io_functions_t'
  cookie_io_functions_t stub = { _iord, _iowr, NULL, NULL };
  ^
```

The same unknown type error occurs when trying to compile `examples/stm32/f4/stm32f429i-discovery/lcd-dma`.

I'm using arm-none-eabi-gcc version 5.3.0 from [this package](https://www.archlinux.org/packages/community/x86_64/arm-none-eabi-gcc/) but the same error occurs with version 5.3.1 20160307 from the [toolchain launchpad distribution](https://launchpad.net/gcc-arm-embedded).

I also have the [arm-non-eabi-newlib](https://www.archlinux.org/packages/community/any/arm-none-eabi-newlib/) package installed, which places its files under `/usr/arm-non-eabi`, but he launchpad toolchain comes with its own copy of newlib too, AFAICS.

What am I missing?

karlp, 2016-05-02:
nothing, it's some 5.3 breakage.  We'll probably have to chagne or drop that example.  what do you need LC_ALL=C for?

SpotlightKid, 2016-05-03:
@karlp Just so that the ouput I paste here is in English.

karlp, 2016-06-17:
fixed in a5641ab

closed 2016-06-17 by karlp