libopencm3/libopencm3-examples #140: Question: prescaler for timers.

2garryn, 2016-07-31:
Hello,
My board is based on MCU stm32f103c8t6.
I am learning and playing with examples and get some misunderstanding with setting prescaler for timers.
I use TIM4, it is connected to APB1 and has frequency 36Mhz for me. 
I want counter to be updated every 1ms. 
What prescaler should I use: 36000 or 35999? 
I have checked examples and here I see 36000:
[https://github.com/libopencm3/libopencm3-examples/blob/master/examples/stm32/f1/stm32-h103/timer/timer.c#L91](https://github.com/libopencm3/libopencm3-examples/blob/master/examples/stm32/f1/stm32-h103/timer/timer.c#L91)

But here prescaler is decremented by 1 :
[https://github.com/libopencm3/libopencm3-examples/blob/master/examples/stm32/l1/stm32l-discovery/button-irq-printf/main.c#L142](https://github.com/libopencm3/libopencm3-examples/blob/master/examples/stm32/l1/stm32l-discovery/button-irq-printf/main.c#L142)

According to reference manual, real prescaler should be decremented:

> Bits15:0 PSC[15:0]:Prescalervalue
The counter clock frequency CK_CNT is equal to fCK_PSC / (PSC[15:0] + 1).

Is it error in examples or libopencm3 decrements prescaler before setting it to register?

Thank you,
Artem

2garryn, 2016-08-05:
Ping anyone

ChuckM, 2016-08-05:
@2garryn I don't use the f103 much but I'm generally in favor of following the reference manuals. Since this is an educational exercise, have you tried toggling a GPIO pin every time the timer fires and measuring the period on an oscilloscope?

karlp, 2016-08-05:
there's a ticket on the library to move the -1 into the library, it's a documentation call really.  The demo that just sets it to 36000 is "wrong" but it's "good enough" for most people.

2garryn, 2016-08-05:
@karlp  
Thank you for explanation. It is very helpful.

@ChuckM No, I didn't :( but will do

closed 2016-08-05 by 2garryn