libopencm3/libopencm3-examples #153: usb_cdcacm wont compile on stm23f4-discovery

Split7fire, 2017-03-27:
Trying to compile in Platformio, and got this:
`src/usbcdc.cpp:146:3: error: elements of array 'const usb_interface ifaces []' have incomplete type
} };
^
src/usbcdc.cpp:146:3: error: storage size of 'ifaces' isn't known
src/usbcdc.cpp: In function 'void cdcacm_set_config(usbd_device*, uint16_t)':

src/usbcdc.cpp:222:27: error: invalid conversion from 'int (*)(usbd_device*, usb_setup_data*, uint8_t**, uint16_t*, void (**)(usbd
_device*, usb_setup_data*)) {aka int (*)(_usbd_device*, usb_setup_data*, unsigned char**, short unsigned int*, void (**)(_usbd_dev
ice*, usb_setup_data*))}' to 'usbd_control_callback {aka int (*)(_usbd_device*, usb_setup_data*, unsigned char**, short unsigned i
nt*, int (**)(_usbd_device*, usb_setup_data*))}' [-fpermissive]
cdcacm_control_request);
^
In file included from src/usbcdc.cpp:23:0:
/home/vlad/.platformio/packages/framework-libopencm3/include/libopencm3/usb/usbd.h:94:12: error:   initializing argument 4 of 'int
 usbd_register_control_callback(usbd_device*, uint8_t, uint8_t, usbd_control_callback)' [-fpermissive]
extern int usbd_register_control_callback(usbd_device *usbd_dev, uint8_t type,
^
Compiling .pioenvs/disco_f407vg/FrameworkLibOpenCM3/lib/stm32/common/adc_common_v1.o`

Any help?

karlp, 2017-03-27:
you have a c++ problem I presume.  Possibly you have a platformio problem, not sure. The examples as shipped compile just fine.  Your log example here hasn't formatted correctly either.

karlp, 2017-03-27:
ok, we can fix this in the library code, https://github.com/libopencm3/libopencm3/issues/762

closed 2017-03-27 by karlp

zyp, 2017-03-27:
This is one C/C++ incompatibility and one case of C++ being stricter than C at type checking.

The first problem warrants a libopencm3 fix, but as a temporary workaround you can replace ```static const struct usb_interface ifaces[] = {{``` with ```static const struct usb_config_descriptor::usb_interface ifaces[] = {{``` on line 140.

The second problem is an error in the example. To fix this you should replace ```void (**complete)(usbd_device *usbd_dev, struct usb_setup_data *req))``` with ```int (**complete)(usbd_device *usbd_dev, struct usb_setup_data *req))``` on line 172.

@karlp mentioned in IRC that there might already be an existing PR for the second problem.