libopencm3/libopencm3-examples #163: Makefile appends - to $(PREFIX)

saleemrashid, 2017-12-31:
The status quo is to use, for example, `$(PREFIX)gcc` instead of `$(PREFIX)-gcc`. (i.e. `PREFIX=arm-none-eabi-` instead of `PREFIX=arm-none-eabi`)

This makes it incompatible with projects that use the former, as libopencm3 will use `arm-none-eabi--gcc` instead.

karlp, 2018-01-01:
example project you're having a hard time with?  I've seen both styles used, because the "arm-none-eabi" is the host triple.  It's simply a matter of style

Further, you' filed this against the examples, which aren't shared with anaything.

closed 2018-01-01 by karlp

saleemrashid, 2018-01-01:
@karlp Oops, I was definitely trying to file this against the main project and have no idea how I filed it against the examples! Shall I refile against the main project?

The project I'm building is https://github.com/trezor/trezor-mcu, but that's not the main issue as I can change that.

The main issue is that I have some compilers I want to use that don't have a triple prefix (e.g. are just called `gcc`) and it would be good to do `PREFIX=/path/to/compiler/bin/` rather than create symlinks.