libopencm3/libopencm3-examples #166: bad examples

karlp, 2018-01-16:
https://github.com/libopencm3/libopencm3-examples/blob/master/examples/stm32/f1/stm32-h103/exti_rising_falling/exti_rising_falling.c

does mention the AFIO requirements if you happen to want to change to another port for the EXTI

karlp, 2019-08-18:
https://github.com/libopencm3/libopencm3-examples/blob/master/examples/stm32/f4/stm32f4-discovery/tick_blink/tick_blink.c#L42 uses an implementation that doesn't handle wrapping correctlky too

ChuckM, 2019-08-19:
```
adc.c: In function 'clock_setup':
adc.c:135:23: error: 'rcc_hsi_8mhz' undeclared (first use in this function); did you mean 'rcc_osc_off'?
  rcc_clock_setup_hsi(&rcc_hsi_8mhz[RCC_CLOCK_64MHZ]);
                       ^~~~~~~~~~~~
                       rcc_osc_off
adc.c:135:23: note: each undeclared identifier is reported only once for each function it appears in
adc.c:135:36: error: 'RCC_CLOCK_64MHZ' undeclared (first use in this function); did you mean 'RCC_CLOCK_HSI_64MHZ'?
  rcc_clock_setup_hsi(&rcc_hsi_8mhz[RCC_CLOCK_64MHZ]);
                                    ^~~~~~~~~~~~~~~
                                    RCC_CLOCK_HSI_64MHZ
../../../../rules.mk:201: recipe for target 'adc.o' failed
make[1]: *** [adc.o] Error 1
Makefile:71: recipe for target 'examples/stm32/f3/stm32f3-discovery/adc/' failed
make: *** [examples/stm32/f3/stm32f3-discovery/adc/] Error 2
```
Fails to build

karlp, 2019-08-19:
Works with current code as is in repo? had you updated the library ahead or something?  (Also, that's just straight up breakage.  this ticket is more for examples that might work, but aren't good examples to copy)

karlp, 2019-08-19:
I also pulled the library to latest and it still builds fine, are you sure you're looking at latest code?

ChuckM, 2019-08-19:
Ok, I remoted the libopencm3 library to upstream, fetched it and pulled master on the sub module to latest. That got rid of that build error but still has a warning about flash_erase_page() on `usbdfu_getstatus_complete` about fallthrough of the case statement. 

And I submitted a PR with a fix for the msleep issue in tick_blink

karlp, 2019-08-20:
yeah, the fallthrough warnings are a different matter.  In one case we even had a comment saying, "fall through" but it was't the style that gcc recognized :)