libopencm3/libopencm3-examples #167: Some examples use deprecated defines.

C47D, 2018-01-19:
Hi,

Both usart and the adc examples on the stm32f0 and the i2c example of the stm32f3 devices are still using deprecated defines (based on libopencm3 head, commit 3922cc7).

### usart and adc examples on stm32f0.
This examples use the following:

`usart_set_stopbits(USART1, USART_CR2_STOP_1_0BIT);`

USART_CR2_STOP_1_0BIT it's only used on this examples, all other usart examples use one of the following defines (usart_cr2_stopbits):
#define USART_STOPBITS_1		USART_CR2_STOPBITS_1   /* 1 stop bit */
#define USART_STOPBITS_0_5	USART_CR2_STOPBITS_0_5 /* .5 stop bit */
#define USART_STOPBITS_2		USART_CR2_STOPBITS_2   /* 2 stop bits */
#define USART_STOPBITS_1_5	USART_CR2_STOPBITS_1_5 /* 1.5 stop bit*/

So the following fix it:
`usart_set_stopbits(USART1, USART_STOPBITS_1);`

### i2c example on stm32f3.
I2C_CR1_DNF_DISABLED is used and it was removed in commit cb2eff (October 26, 2017). using 0 is fine in this case.

Error:
`i2c_set_digital_filter(I2C1, I2C_CR1_DNF_DISABLED);`

Solved with:
`i2c_set_digital_filter(I2C1, 0);`

Regards,
Carlos

karlp, 2018-01-19:
Yes, this is why the library submodule reference hasn't been updated in the examples repo yet.  They should all get updated together.

C47D, 2018-01-23:
Ok, np, i guess this will remain open until the submodule gets updated, any date for that?

karlp, 2018-03-03:
this has all been updated now.

closed 2018-03-03 by karlp