libopencm3/libopencm3-examples #174: stm32f3-discovery I2C example not working

Palmitoxico, 2018-03-10:
Hi,

I've compiled and flashed the i2c example provided on ```examples/stm32/f3/stm32f3-discovery/i2c``` but  "Hello, we're running" is printed on serial once the micro is reseted and nothing happens.

I've debugged the problem and found out that it gets stuck inside a loop in the i2c_transfer7 function:
 libopencm3/lib/stm32/common/i2c_common_v2.c lines 404 - 411:
```
	bool wait = true;
	while (wait) {
		if (i2c_transmit_int_status(i2c)) {
			wait = false;
		}
		while (i2c_nack(i2c)); /* FIXME Some error */
	}
	i2c_send_data(i2c, *w++);
```
The ```i2c_transmit_int_status(i2c)``` returns always 0, so the ```while(wait) {``` loop never exits. The ```while (i2c_nack(i2c));``` line never gets stuck, so it isn't the culprit. 

I've compiled with arm-none-eabi-gcc 7.3.0 on Archlinux.

karlp, 2018-03-10:
Thanks for the report.  https://github.com/karlp/libopencm3-tests/blob/master/tests/i2c-master/main-stm32f072-disco.c is probably a more up to ddate example of working with the i2c-v2 peripheral on f3.

Palmitoxico, 2018-03-11:
I'll try this tomorrow. Thanks.

Sghazzawi, 2023-08-12:
I am running into the same issue.  I have the Discovery board with the STM32F303.  The example linked above is for an SHT21 sensor (which is not on this board).  Are there any working examples of using I2C on STM32F303/Discovery?