libopencm3/libopencm3-examples #176: Warning while building cdcacm.c

Boldie, 2018-03-21:
While building the examples, the following new warning is created with the newest libopencm3 master:

```
libopencm3-examples/examples/stm32/f1/stm32-h103/usb_cdcacm$ make
cdcacm.c: In function 'cdcacm_set_config':
cdcacm.c:232:5: warning: passing argument 4 of 'usbd_register_control_callback' from incompatible pointer type
     cdcacm_control_request);
     ^
In file included from cdcacm.c:23:0:
../../../../../libopencm3//include/libopencm3/usb/usbd.h:130:12: note: expected 'usbd_control_callback' but argument is of type 'enum usbd_request_return_codes (*)(struct usbd_device *, struct usb_setup_data *, uint8_t **, uint16_t *, void (**)(struct usbd_device *, struct usb_setup_data *))'
 extern int usbd_register_control_callback(usbd_device *usbd_dev, uint8_t type,
            ^
```

karlp, 2018-03-21:
YOu mean, not with the specific revision that the git submodule is set to, but by explicitly updating the library yourself?  Because that's expected.  That's why git submodules include version information

karlp, 2018-03-21:
failing that, are both your repos up to date?  Even updating the submodule to latest is still warning clean for me.  The warnings you are seeing should have all been resolved via 301b4f91942cb3df5a6c1458f74fdbebc0c12548

Boldie, 2018-03-21:
I did a git pull, shouldn't this also update the submodule to the pinned revision?

Boldie, 2018-03-21:
Ok, got it. *** submodules. Not very intuitive to use.

closed 2018-03-21 by Boldie

karlp, 2018-03-22:
there's a "pull --recurse-submodules" option, not sure if you can set it in gitconfig or not.