libopencm3/libopencm3-examples #178: stm32f4-discovery USB examples don't work

felipebalbi, 2018-04-22:
Hi,

I have an stm32f4-discovery board labeled MB997D which comes with STM32F407VG. I tested every example on the stm32f4-discovery directory and they all work, except for the USB examples. I didn't get my scope hooked up yet, but if I were to guess, Data Pull up isn't being connected.

I suppose the reason for data pullup to not being connected, even though we clear bit 1 in DCTL, is because the USB IP doesn't realize that VBUS is above 4.4V. Going through the datasheet, I found out that PA9 is not, anymore, used as OTG_FS_VBUS. In fact, AF10 is not a valid mux mode for GPIOA_10.

Any hints about what to do here? I actually have this same failure on two different boards (stm32f4-discovery and stm32-h405 from Olimex).

zyp, 2018-04-22:
OTG_FS_VBUS being an AF was always a misconception. PA9 should be configured in analog mode when used to sense VBUS, and as such is not listed as an AF anymore.

I don't really think this is the root of your problem, but if it is, it should be fairly easy to diagnose with the debugger:

If the USB core is otherwise configured correctly, the BSVLD bit in OTG_FS_GOTGCTL will indicate whether VBUS is present or not. When BSVLD is set, the DP pullup is enabled.

You can disable VBUS sensing by setting the NOVBUSSENS bit in OTG_FS_GCCFG. Setting it will make the USB core always assume VBUS is present.

ChuckM, 2018-04-22:
Exactly right @felipebalbi the difference is the treatment of bit 21. New versions of the USB hardware need it set in order to work.

zyp, 2018-04-22:
@ChuckM Do you have a reference for that assertion?

I'm not aware of VBUS sensing not being available in newer parts, and if it's supposed to work, disabling it is just hiding the issue, not solving it.

karlp, 2018-04-22:
Is it just that https://github.com/libopencm3/libopencm3/blob/master/lib/usb/usb_f107.c#L65-L71 is missing from the usb_f207 file?  So this is only noticed on boards that have the OTG_HS periph wired up?

felipebalbi, 2018-04-23:
@karlp I can test that out later. But this begs the question: if they have the same USB IP, why two different drivers?

zyp, 2018-04-23:
As far as I can see, there's no relevant changes in the f4 disco schematics, it's still connected to OTG_FS.

Furthermore, latest revision of RM0090 still lists bit 21 of OTG_FS_GCCFG as being NOVBUSSENS, not VBDEN. I guess this could be checked by reading out OTG_FS_CID with the debugger.

Try checking the level on PA9 with a scope, check that you actually have 5V there.

karlp, 2018-06-08:
Any further updates here?

karlp, 2018-08-29:
No updates, closing.

closed 2018-08-29 by karlp

44670, 2019-04-07:
The stm32f4 examples is not workable because the vbus sensing is not proply configured.

Vbus sensing is enabled by default in the driver code.
https://github.com/libopencm3/libopencm3/blob/ff9664389bfb560145aa3b36ad8368900c57c3d8/lib/usb/usb_f107.c#L70

In this case, the GPIO PA9 must be configured as vbus sensing function, however, only PA11, PA12 is configured in all of the usb examples in stm32f4 directory. 
https://github.com/libopencm3/libopencm3-examples/blob/f9713ff86508a522a2c02bde947b57a871bc34bc/examples/stm32/f4/stm32f4-discovery/usb_cdcacm/cdcacm.c#L235

This problem could be solved by properly configure PA9 pin or disable vbus sensing. By adding following code to disable vbus sensing, the example is workable for me.

OTG_FS_GCCFG |= OTG_GCCFG_NOVBUSSENS;
OTG_FS_GCCFG &= ~OTG_GCCFG_VBUSASEN;
OTG_FS_GCCFG &= ~OTG_GCCFG_VBUSBSEN;

zyp, 2019-04-07:
> In this case, the GPIO PA9 must be configured as vbus sensing function…

No, like I wrote in my first comment on this issue, OTG_FS_VBUS is not an AF and PA9 should for that reason not be configured to AF mode. The VBUS sense circuitry is directly connected to the pin and will sense it regardless of what mode it is in.

According to documentation, AF10 is not a valid AF for PA9, and as far as I can remember in practice will sink current. In other words, if you configure it like that, you'll be feeding 5V into a pin that's trying to output low. VBUS can generally source plenty of current to overcome the drive strength of the output, so it will still appear to work normally, but the chip will get noticeably warmer from all the power going into that pin.

pqc-lac, 2020-04-25:
I have this problem too. This problem can be solved by just  update the version of stink. With the old version of stink, the disc1 board can only be powered by the pc USB port. If you use other external USB port directly from adapter, then the stm32f4 mcc will not be powered. The new version of the stink firmware solve this bug.

julian1, 2020-07-11:
Following the discussion, I got the usb example on my stm32f4 discovery board to work with, 

```
vim  ./libopencm3/lib/usb/usb_f107.c
70     // OTG_FS_GCCFG |= OTG_GCCFG_VBUSBSEN | OTG_GCCFG_PWRDWN;
71     OTG_FS_GCCFG |= OTG_GCCFG_NOVBUSSENS | OTG_GCCFG_PWRDWN; 
``` 

I would like to encourage re-opening the issue, to provide more visibility - even if it's not clear how to proceed with a  generic patch to cover all use-cases. 

It is quite possible to waste a lot of time, assuming user error, and buying and testing different boards, before discovering the solution.

davidgfnet, 2020-07-14:
My god took me an hour to find this :P  
Would it be at least useful to update the examples repo with some comments on this? The fact that PA9 does not require any initialization confuses the user in the examples, wrongly assuming that PA9 is not uses at all!

tormodvolden, 2021-12-01:
Some comments to the original post:
- "because the USB IP doesn't realize that VBUS is above 4.4V" - I don't think the threshold is 4.4V. If you go through the reference manual and the USB OTG specification for "b-session valid" it should be between 0.8 V and 4.0 V. ST recommends in AN4879 (FAQ section or [here](https://st.force.com/community/s/article/FAQ-Management-of-VBUS-sensing-for-USB-device-design)) applying 0.7 * Vcc for their products (so 2.3V in this context) and not above 4.0V (to avoid breaking the Vcc+4V limitation if the chip is unpowered) and a super-user at the ST forums [experimentally determined](https://community.st.com/s/question/0D50X00009XkaC1SAJ/pa9-as-vbus-sense-of-usbotgfs-stm32f407) the threshold to be around 1.5V (like a Schmidt-trigger input) at 3.3V Vcc. I have been checking this because some boards have PA9 connected to Vcc instead of VBUS, which for a bus-powered device should therefore be fine.

-  "this same failure on two different boards (stm32f4-discovery and stm32-h405 from Olimex)" - When it comes to the Olimex STM32-H405 the [schematics](https://www.olimex.com/Products/ARM/ST/STM32-H405/) show that PA9 is not connected to anything except header EXT1-4. It can easily be bridged to USB VBUS on EXT2-23 I guess, in case the firmware doesn't disable VBUSSENSE. This board also have the external D+ pullup circuitry needed for STM32F103, but I guess it won't help for the F407 USB if it is "stuck" on missing VBUS.

What remains unclear here is why newer STM32F4-Discovery boards don't work. In latest revisions of its reference manual, ST has removed the schematics. Can someone please verify (ideally providing schematics) that USB VBUS is no longer connected to PA9 like it was in earlier board revisions?

E.g the STM32Duino project has gone as far as [disabling VBUSSENSE by default](https://github.com/stm32duino/Arduino_Core_STM32/pull/954) in their library, letting users enable it if PA9 has been hooked up correctly. This would be a good solution for libopencm3 also IMO.

manuelbl, 2021-12-01:
The current USB implementation in libopencm3 contains several bugs. The way VBUS sensing is implemented is close to a bug (unfortunate and undocumented default setting that cannot be changed with libopencm3 functions). Likely many people have wasted hours until they identified the root issue, or have never succeeded and turned away from libopencm3.

The main problem however is that the project seems to be unable to fix them. The last time a USB bug was "fixed" was more than 3 years ago (https://github.com/libopencm3/libopencm3/commit/26ab78a710b4cb7abcace769768c4781332b17ba). And it broke more than it fixed and should be undone. There are several PRs to address some of the bugs, with no progress.

I don't have any real insights into the project. I can see that @karlp has an active role. But he seems to be the only one, and he does not accept any USB changes - for reasons unknown to me. I also understand that there was a good system for integration testing the USB code. But - as far as I can tell - it's broken now.

I'm sorry to say so, but if you need USB functionality you are probably better off writing your own USB extension for libopencm3 or move to a different framework altogether.

julian1, 2021-12-01:
Maybe there is an assumption the issue pertains to just vbus sensing, when the default and undocumented behavior is to break all usb functionality for f4. It is easy to workaround - and usb functionality then works great from my experience. Tthe problem is one of visibility and having the wrong expectations about code presented as working examples.