libopencm3/libopencm3-examples #180: Old vs new style ld scripts

diegoherranz, 2018-05-19:
There are two ways of dealing with linker scripts (very roughly):
- Old style: static linker script. Example: defining LDSCRIPT=[stm32-h103.ld](https://github.com/libopencm3/libopencm3-examples/blob/master/examples/stm32/f1/stm32-h103/stm32-h103.ld) in the Makefile.
- New style: linker dynamically generated. Example: defining device=STM32F103RBT6 and including [genlink-config.mk](https://github.com/libopencm3/libopencm3/blob/master/mk/genlink-config.mk) in the Makefile

libopencm3-examples seems to support both on its main Makefile ([rules.mk](https://github.com/libopencm3/libopencm3-examples/blob/master/examples/rules.mk#L82)) but most of the examples (all?) use the old style:
- First of all, I guess the new style is the preferred way in libopencm3?
- If so, why are the examples using the old way? Simply because no one has done? Or is it intentional? If the former, I could help porting some examples to the new style.

Thanks!

karlp, 2018-05-19:
yes, mostly because it's just "churn" to change them.  it's also fairly noticeably slower to build all the examples with with generated linker scripts.  (I've done it before, with the old awk based one, I've not tested the new python one)

You can use either.  The generator is less maintenance on a huge pile of linker scripts in the library, which is good for the maintainers, but it makes ~zero difference to the end user which style you use.  Using a fixed linker script gives you more options for your own build environment, the dynamic linker generator is only supported as some makefile stubs to include.  

As far as "preferred" goes, you can assume that the generator is preferred, in as much as we won't be merging more per device linker scripts in the main library.

closed 2018-05-19 by karlp

diegoherranz, 2018-05-20:
Yes, I think it's easier in terms of maintaining it. For example, if there was any typo in linker sections, for the automated way, it only needs to be fixed on the main library's devices.data.

Anyway, I understand it's a burden to modify every example to use the new way.

Thanks.