libopencm3/libopencm3-examples #182: [STM32F429I-DISCO] lcd-dma example does not work with libopencm3 referenced library

GaoTamanrasset, 2018-07-03:
lcd-dma example (https://github.com/libopencm3/libopencm3-examples/tree/master/examples/stm32/f4/stm32f429i-discovery/lcd-dma) on stm32f429i discovery board, not work with the  referenced libopencm3 version (dc64929).

ChuckM, 2018-07-03:
What does is do, or not do, that you would expect? It seems to compile fine.

GaoTamanrasset, 2018-07-03:
Yes, it compile fine.

But, I expect this : 
[image]

and I have this : 
[image]

First picture is lcd-dma example I have compiled from pull request 141:
`git clone https://github.com/libopencm3/libopencm3-examples.git`
`cd libopencm3-examples/`
`git fetch origin pull/141/head:testPRlcd`
`git checkout testPRlcd`
`git submodule init`
`git submodule update`
`make`
`cd examples/stm32/f4/stm32f429i-discovery/lcd-dma/`
`make cd-dma.stlink-flash̀`

lcd-dma source files are identical in HEAD branch et in the pull request 141. 
Just libopencm3 versions changes.
An idea ?

PS : 
`git rev-parse HEAD`
in libopencm3 directory from the pull request 141 branch : fb410a403db7298f97fb0120bb93f822ec0e2aa4
in libopencm3 directory from the HEAD branch : 
dc64929b78e5e0ba3be31577a0804f95c92b52cd

ChuckM, 2018-07-03:
From the pictures it looks like the LTDC peripheral and the RCC (clock) are not set up correctly. I would look at the RCC_PLLCFGR register on both builds and the LTDC registers for differences.

GaoTamanrasset, 2018-07-03:
In gdb : 
`set {int}0x40023804 = 0x07405408`
`continue`

And I still have the same problem.

GaoTamanrasset, 2018-07-03:
And : 

$ diff LCD1.txt LCD2.txt
3c3
< 0x40016820:	0x00000020	0x00000000	0x00000000	0x00590000
---
> 0x40016820:	0x00000020	0x00000000	0x00000000	0x00000067
5c5
< 0x40016840:	0x00000000	0x000b008d	0x00000000	0x00000000
---
> 0x40016840:	0x00000000	0x00300012	0x00000000	0x00000000
17,18c17,18
< 0x40016900:	0xff50a075	0x00000001	0x01050086	0x00df0060
< 0x40016910:	0x00000000	0x00000004	0x00000045	0x00000000
---
> 0x40016900:	0xff50a075	0x00000001	0x00f40075	0x010b008c
> 0x40016910:	0x00000000	0x00000004	0x000000cd	0x00000000

LCD1 : bad binary
LCD2 : good binary

ChuckM, 2018-07-03:
That upper bits is the PLLR field that is used on the F4x9 and F4x6 series. It sets the clock source for the DSI peripheral and LTDC dot clock. I don't think it is reserved on the '429, I think it really is needed. The differences in the LTDC registers will take a bit of looking. I know I had to rewhack things on my 469 repo to get it to work with the current LOC3. 

In the mean time you might want to decode the registers and see what is being set there to figure out if your DMA is running on the correct clock.

GaoTamanrasset, 2018-07-04:
First check : 

PLLCFGR register (in RM0090 §2.3 Table 1. RCC boundary address is 0x4002 3800 - 0x4002 3BFF and § 
7.3.24 Table 34 address offset of RCC_PLLCFGR register is 0x04) : **0x40023804**

Value read in the correct correct :    0x07405408
And in the incorrect configuration : 0x27405408

Difference is bit 29. In RM0090 § 7.3.2 : "Bits 31:28 Reserved, must be kept at reset value."

Is my analyse wrong ?

ChuckM, 2018-07-05:
I have created a separate repo for this to figure out what broke. So far 11 of 21 of the examples work, 3 of them don't work (lcd-dma, dac-dma, and dac). I should be able to bisect the loc3 repo to find which commit broke things.

ChuckM, 2018-07-06:
The separate repo is here :  https://github.com/ChuckM/stm32f429-discovery-examples

ChuckM, 2018-07-06:
The error is this line of lcd-dma.c
```
-       RCC_DCKCFGR |= RCC_DCKCFGR_PLLSAIDIVR_DIVR_8;
+       RCC_DCKCFGR |= (RCC_DCKCFGR_PLLSAIDIVR_DIVR_8 << RCC_DCKCFGR_PLLSAIDIVR_SHIFT);
``` 

Change that and the example works again, thanks to zyp who identified where it would be once I found which commit it occured in.

GaoTamanrasset, 2018-07-07:
Thanks ! It works fine.

closed 2018-07-07 by GaoTamanrasset