libopencm3/libopencm3-examples #183: Reset vector question

diegoherranz, 2018-07-10:
I'm trying to understand the fundamentals of the STM32/CortexM3 microcontrollers so I was inspecting the elf files generated for these examples.

Namely I was inspecting the elf compiled from the [miniblik for stm32-h103](https://github.com/libopencm3/libopencm3-examples/tree/master/examples/stm32/f1/stm32-h103/miniblink) [excerpt]:

```
$ arm-none-eabi-objdump -d miniblink.elf 

miniblink.elf:     file format elf32-littlearm

Disassembly of section .text:

08000000 <vector_table>:
 8000000:	00 50 00 20 41 02 00 08 3f 02 00 08 3d 02 00 08     .P. A...?...=...
 8000010:	3d 02 00 08 3d 02 00 08 3d 02 00 08 00 00 00 00     =...=...=.......
	...

08000150 <main>:
 8000150:	b537      	push	{r0, r1, r2, r4, r5, lr}
 8000152:	2400      	movs	r4, #0
	...

08000240 <reset_handler>:
 8000240:	2200      	movs	r2, #0
 8000242:	4c18      	ldr	r4, [pc, #96]	; (80002a4 <reset_handler+0x64>)
 8000244:	4918      	ldr	r1, [pc, #96]	; (80002a8 <reset_handler+0x68>)
 8000246:	4819      	ldr	r0, [pc, #100]	; (80002ac <reset_handler+0x6c>)
	...
```

I was starting with the vector table. You can see the reset vector, which starts at offset 0x04, is 0x08000241 (little endian)
-  8000000:	00 50 00 20 **41 02 00 08** 3f 02 00 08 3d 02 00 08     .P. A...?...=...

Why is that one address higher than the reset_handler routine? I was expecting those two addresses to be the same.

I'm guessing it is correct because it all seems to work but I can't understand it. I've been reading ARM and ST documentation but can't find any explanation.

Many thanks!

karlp, 2018-07-12:
I presume you're seeing an artifact of thumb instructions always having the lower bit set, to distinguish them from arm instructions.  (16vs32bit wide instructions)

closed 2018-07-12 by karlp

diegoherranz, 2018-07-14:
Yes, that's what it is.
I've eventually found it on the [Cortex-M3 Devices Generic User Guide](http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.dui0552a/BABIFJFG.html).

Many thanks for your help.