libopencm3/libopencm3-examples #191: make error 'cannot find -lopencm3_stm32f7'

danbates2, 2019-01-24:
Here's the last few lines of make.. 
Any thoughts?

```
  GENHDR  include/libopencm3/sam/4l/irq.json
  GENHDR  include/libopencm3/sam/d/irq.json
  GENHDR  include/libopencm3/vf6xx/irq.json
rm -f .stamp_failure_lib_stm32_f0
  BUILD   lib/stm32/f0
/Applications/Xcode.app/Contents/Developer/usr/bin/make --directory=lib/stm32/f0 SRCLIBDIR="/Users/admin/Documents/github/libopencm3-miniblink/libopencm3/lib" || \
		echo "Failure building: lib/stm32/f0: code: $?" > .stamp_failure_lib_stm32_f0
make[2]: Nothing to be done for `all'.
rm -f .stamp_failure_lib_stm32_f1
  BUILD   lib/stm32/f1
/Applications/Xcode.app/Contents/Developer/usr/bin/make --directory=lib/stm32/f1 SRCLIBDIR="/Users/admin/Documents/github/libopencm3-miniblink/libopencm3/lib" || \
		echo "Failure building: lib/stm32/f1: code: $?" > .stamp_failure_lib_stm32_f1
make[2]: Nothing to be done for `all'.
rm -f .stamp_failure_lib_stm32_f2
  BUILD   lib/stm32/f2
/Applications/Xcode.app/Contents/Developer/usr/bin/make --directory=lib/stm32/f2 SRCLIBDIR="/Users/admin/Documents/github/libopencm3-miniblink/libopencm3/lib" || \
		echo "Failure building: lib/stm32/f2: code: $?" > .stamp_failure_lib_stm32_f2
make[2]: Nothing to be done for `all'.
rm -f .stamp_failure_lib_stm32_f3
  BUILD   lib/stm32/f3
/Applications/Xcode.app/Contents/Developer/usr/bin/make --directory=lib/stm32/f3 SRCLIBDIR="/Users/admin/Documents/github/libopencm3-miniblink/libopencm3/lib" || \
		echo "Failure building: lib/stm32/f3: code: $?" > .stamp_failure_lib_stm32_f3
make[2]: Nothing to be done for `all'.
rm -f .stamp_failure_lib_stm32_f4
  BUILD   lib/stm32/f4
/Applications/Xcode.app/Contents/Developer/usr/bin/make --directory=lib/stm32/f4 SRCLIBDIR="/Users/admin/Documents/github/libopencm3-miniblink/libopencm3/lib" || \
		echo "Failure building: lib/stm32/f4: code: $?" > .stamp_failure_lib_stm32_f4
make[2]: Nothing to be done for `all'.
rm -f .stamp_failure_lib_stm32_f7
  BUILD   lib/stm32/f7
/Applications/Xcode.app/Contents/Developer/usr/bin/make --directory=lib/stm32/f7 SRCLIBDIR="/Users/admin/Documents/github/libopencm3-miniblink/libopencm3/lib" || \
		echo "Failure building: lib/stm32/f7: code: $?" > .stamp_failure_lib_stm32_f7
Makefile:72: Your toolchain doesn't support -mcpu=cortex-m7.
Please use gcc-arm-embedded 4.8 2014q3 or newer. Skipping this sub-library.
rm -f .stamp_failure_lib_stm32_l0
  BUILD   lib/stm32/l0
/Applications/Xcode.app/Contents/Developer/usr/bin/make --directory=lib/stm32/l0 SRCLIBDIR="/Users/admin/Documents/github/libopencm3-miniblink/libopencm3/lib" || \
		echo "Failure building: lib/stm32/l0: code: $?" > .stamp_failure_lib_stm32_l0
make[2]: Nothing to be done for `all'.
rm -f .stamp_failure_lib_stm32_l1
  BUILD   lib/stm32/l1
/Applications/Xcode.app/Contents/Developer/usr/bin/make --directory=lib/stm32/l1 SRCLIBDIR="/Users/admin/Documents/github/libopencm3-miniblink/libopencm3/lib" || \
		echo "Failure building: lib/stm32/l1: code: $?" > .stamp_failure_lib_stm32_l1
make[2]: Nothing to be done for `all'.
rm -f .stamp_failure_lib_stm32_l4
  BUILD   lib/stm32/l4
/Applications/Xcode.app/Contents/Developer/usr/bin/make --directory=lib/stm32/l4 SRCLIBDIR="/Users/admin/Documents/github/libopencm3-miniblink/libopencm3/lib" || \
		echo "Failure building: lib/stm32/l4: code: $?" > .stamp_failure_lib_stm32_l4
make[2]: Nothing to be done for `all'.
rm -f .stamp_failure_lib_lpc13xx
  BUILD   lib/lpc13xx
/Applications/Xcode.app/Contents/Developer/usr/bin/make --directory=lib/lpc13xx SRCLIBDIR="/Users/admin/Documents/github/libopencm3-miniblink/libopencm3/lib" || \
		echo "Failure building: lib/lpc13xx: code: $?" > .stamp_failure_lib_lpc13xx
make[2]: Nothing to be done for `all'.
rm -f .stamp_failure_lib_lpc17xx
  BUILD   lib/lpc17xx
/Applications/Xcode.app/Contents/Developer/usr/bin/make --directory=lib/lpc17xx SRCLIBDIR="/Users/admin/Documents/github/libopencm3-miniblink/libopencm3/lib" || \
		echo "Failure building: lib/lpc17xx: code: $?" > .stamp_failure_lib_lpc17xx
make[2]: Nothing to be done for `all'.
rm -f .stamp_failure_lib_lpc43xx_m4
  BUILD   lib/lpc43xx/m4
/Applications/Xcode.app/Contents/Developer/usr/bin/make --directory=lib/lpc43xx/m4 SRCLIBDIR="/Users/admin/Documents/github/libopencm3-miniblink/libopencm3/lib" || \
		echo "Failure building: lib/lpc43xx/m4: code: $?" > .stamp_failure_lib_lpc43xx_m4
make[2]: Nothing to be done for `all'.
rm -f .stamp_failure_lib_lpc43xx_m0
  BUILD   lib/lpc43xx/m0
[... 79 lines cut ...]
  BUILD   lib/sam/4l
/Applications/Xcode.app/Contents/Developer/usr/bin/make --directory=lib/sam/4l SRCLIBDIR="/Users/admin/Documents/github/libopencm3-miniblink/libopencm3/lib" || \
		echo "Failure building: lib/sam/4l: code: $?" > .stamp_failure_lib_sam_4l
make[2]: Nothing to be done for `all'.
rm -f .stamp_failure_lib_sam_d
  BUILD   lib/sam/d
/Applications/Xcode.app/Contents/Developer/usr/bin/make --directory=lib/sam/d SRCLIBDIR="/Users/admin/Documents/github/libopencm3-miniblink/libopencm3/lib" || \
		echo "Failure building: lib/sam/d: code: $?" > .stamp_failure_lib_sam_d
make[2]: Nothing to be done for `all'.
rm -f .stamp_failure_lib_vf6xx
  BUILD   lib/vf6xx
/Applications/Xcode.app/Contents/Developer/usr/bin/make --directory=lib/vf6xx SRCLIBDIR="/Users/admin/Documents/github/libopencm3-miniblink/libopencm3/lib" || \
		echo "Failure building: lib/vf6xx: code: $?" > .stamp_failure_lib_vf6xx
make[2]: Nothing to be done for `all'.
rm -f .stamp_failure_tld
for failure in .stamp_failure_*; do \
		[ -f $failure ] && cat $failure >> .stamp_failure_tld || true; \
	done;
[ -f .stamp_failure_tld ] && cat .stamp_failure_tld && exit 1 || true;
  stm32f7 -> Creating bin/stm32/stm32f746g-disco.elf
arm-none-eabi-gcc -DRCC_LED1=RCC_GPIOI -DPORT_LED1=GPIOI -DPIN_LED1=GPIO1 --static -nostartfiles -std=c11 -g3 -Os -fno-common -ffunction-sections -fdata-sections -I./libopencm3/include -L./libopencm3/lib -mcpu=cortex-m7 -mthumb -mfloat-abi=hard -mfpu=fpv5-sp-d16 -DSTM32F7 -DLITTLE_BIT=800000 -Wl,--start-group -lc -lgcc -lnosys -Wl,--end-group template_stm32.c -T ld.stm32.basic -lopencm3_stm32f7 -o bin/stm32/stm32f746g-disco.elf
/usr/local/gcc-arm-none-eabi-8-2018-q4-major/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld: cannot find -lopencm3_stm32f7
collect2: error: ld returned 1 exit status
make: *** [bin/stm32/stm32f746g-disco.elf] Error 1
```

karlp, 2019-01-24:
Did you run make in the root directory first?

danbates2, 2019-01-24:
Grr.. messy day, I meant to create this issue in the libopencm3-miniblink repo.
https://github.com/libopencm3/libopencm3-miniblink
Here's a non-verbose version of the error.

```
  BUILD   lib/stm32/f4
  BUILD   lib/stm32/f7
Makefile:72: Your toolchain doesn't support -mcpu=cortex-m7.
Please use gcc-arm-embedded 4.8 2014q3 or newer. Skipping this sub-library.
  BUILD   lib/stm32/l0
  BUILD   lib/stm32/l1
  BUILD   lib/stm32/l4
  BUILD   lib/lpc13xx
  BUILD   lib/lpc17xx
  BUILD   lib/lpc43xx/m4
  BUILD   lib/lpc43xx/m0
  BUILD   lib/lm3s
  BUILD   lib/lm4f
  BUILD   lib/msp432/e4
  BUILD   lib/efm32/tg
  BUILD   lib/efm32/g
  BUILD   lib/efm32/lg
  BUILD   lib/efm32/gg
  BUILD   lib/efm32/hg
  BUILD   lib/efm32/wg
  BUILD   lib/efm32/ezr32wg
  BUILD   lib/sam/3a
  BUILD   lib/sam/3n
  BUILD   lib/sam/3s
  BUILD   lib/sam/3u
  BUILD   lib/sam/3x
  BUILD   lib/sam/4l
  BUILD   lib/sam/d
  BUILD   lib/vf6xx
  stm32f7 -> Creating bin/stm32/stm32f746g-disco.elf
/usr/local/gcc-arm-none-eabi-8-2018-q4-major/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld: cannot find -lopencm3_stm32f7
collect2: error: ld returned 1 exit status
make: *** [bin/stm32/stm32f746g-disco.elf] Error 1
AdminNamesMBPro:libopencm3-miniblink admin$ 
```

closed 2019-01-24 by danbates2

karlp, 2019-01-24:
Looks like the cortex-m7 check is failing on your toolchain.  This was actually a "bug" in the library makefiles when newer gcc changed their help output.  We dropped the check, (cortex-m7 support is now widespread) but hte miniblink repo wasn't updated.  It now has been: https://github.com/libopencm3/libopencm3-miniblink/commit/986296b77d8fae8f37cb784541a036c35b796f36

danbates2, 2019-01-24:
That worked.
It all compiles and all .bin files are created. Thanks.