libopencm3/libopencm3-examples #201: libopencm3 make failed

kynguyen98, 2020-02-22:
when I clone the libopencm3-example and do make It gave this 
```
  GENHDR  include/libopencm3/stm32/f0/irq.json
  GENHDR  include/libopencm3/stm32/f1/irq.json
  GENHDR  include/libopencm3/stm32/f2/irq.json
  GENHDR  include/libopencm3/stm32/f3/irq.json
  GENHDR  include/libopencm3/stm32/f4/irq.json
  GENHDR  include/libopencm3/stm32/f7/irq.json
  GENHDR  include/libopencm3/stm32/l0/irq.json
  GENHDR  include/libopencm3/stm32/l1/irq.json
  GENHDR  include/libopencm3/stm32/l4/irq.json
  GENHDR  include/libopencm3/stm32/g0/irq.json
  GENHDR  include/libopencm3/gd32/f1x0/irq.json
  GENHDR  include/libopencm3/lpc13xx/irq.json
  GENHDR  include/libopencm3/lpc17xx/irq.json
  GENHDR  include/libopencm3/lpc43xx/m4/irq.json
  GENHDR  include/libopencm3/lpc43xx/m0/irq.json
  GENHDR  include/libopencm3/lm3s/irq.json
  GENHDR  include/libopencm3/msp432/e4/irq.json
  GENHDR  include/libopencm3/efm32/tg/irq.json
  GENHDR  include/libopencm3/efm32/g/irq.json
  GENHDR  include/libopencm3/efm32/lg/irq.json
  GENHDR  include/libopencm3/efm32/gg/irq.json
  GENHDR  include/libopencm3/efm32/hg/irq.json
  GENHDR  include/libopencm3/efm32/wg/irq.json
  GENHDR  include/libopencm3/efm32/ezr32wg/irq.json
  GENHDR  include/libopencm3/sam/3a/irq.json
  GENHDR  include/libopencm3/sam/3n/irq.json
  GENHDR  include/libopencm3/sam/3s/irq.json
  GENHDR  include/libopencm3/sam/3u/irq.json
  GENHDR  include/libopencm3/sam/3x/irq.json
  GENHDR  include/libopencm3/sam/4l/irq.json
  GENHDR  include/libopencm3/sam/d/irq.json
  GENHDR  include/libopencm3/vf6xx/irq.json
  GENHDR  include/libopencm3/swm050/irq.json
  BUILD   lib/stm32/f0
/bin/sh: -c: line 1: unexpected EOF while looking for matching `"'
/bin/sh: -c: line 2: syntax error: unexpected end of file
make[1]: *** [Makefile:71: lib/stm32/f0] Error 1
make: *** [Makefile:58: lib] Error 2
```
I think there is something wrong with lib/stm32/f0

karlp, 2020-02-22:
I've just pushed an update to the latest library version, which includes a fix for make 4.3 (as used on arch for instance) let me know if that doesn't fix things for you, thanks for the report!

closed 2020-02-22 by karlp

kynguyen98, 2020-02-22:
It worked thanks mate