libopencm3/libopencm3-examples #210: STM32F446 example for usb

riyas-org, 2020-07-31:
Having a hard time to get stm32f446 usb to work with the library. Tried to adapt the cdcacm example for F429 discovery and by changing  OTG_FS_GCCFG &= ~OTG_GCCFG_VBDEN in the usb_f107.c file (to disable vbus detection).  Looking forward for an example code to start with usb devices on 446 (More of a feature request:)

scaldov, 2023-03-03:
The same on stm32f412z

scaldov, 2023-03-04:
correct work on f411:
----------------------
-|6|SRQ/CIDSCHG/PTXFE/EOPF/USBSUSP/ESUSP/NPTXFE/
-|196|SRQ/CIDSCHG/PTXFE/RSTDET/EOPF/ENUMDNE/USBRST/ESUSP/NPTXFE//ED//R/
-|268|PTXFE/EOPF/NPTXFE/RXFLVL/SOF/
|268|PTXFE/EOPF/NPTXFE/RXFLVL/SOF/
|269|PTXFE/IEP/EOPF/NPTXFE/SOF/[80/06/00/40/100/]

|269|PTXFE/EOPF/USBRST/NPTXFE/RXFLVL//R/
|269|PTXFE/NPTXFE/RXFLVL/
-|323|PTXFE/ENUMDNE/NPTXFE//ED/
|393|PTXFE/EOPF/NPTXFE/RXFLVL/SOF/
|393|PTXFE/EOPF/NPTXFE/RXFLVL/SOF/
|393|PTXFE/IEP/EOPF/NPTXFE/SOF/[00/05/00/00/14/]
ADDR 20

-|411|PTXFE/EOPF/NPTXFE/RXFLVL/SOF/
|411|PTXFE/EOPF/NPTXFE/RXFLVL/
|412|PTXFE/IEP/EOPF/NPTXFE/SOF/[80/06/00/12/100/]

|412|PTXFE/EOPF/NPTXFE/RXFLVL/
|412|PTXFE/EOPF/NPTXFE/RXFLVL/
|412|PTXFE/EOPF/NPTXFE/RXFLVL/SOF/
|412|PTXFE/EOPF/NPTXFE/RXFLVL/SOF/
|413|PTXFE/EOPF/NPTXFE/RXFLVL/SOF/
|413|PTXFE/EOPF/NPTXFE/RXFLVL/
|414|PTXFE/EOPF/NPTXFE/RXFLVL/SOF/
|414|PTXFE/EOPF/NPTXFE/RXFLVL/
|415|PTXFE/EOPF/NPTXFE/RXFLVL/SOF/
|415|PTXFE/EOPF/NPTXFE/RXFLVL/SOF/
|415|PTXFE/IEP/EOPF/NPTXFE/SOF/[80/06/00/09/200/]

scaldov, 2023-03-04:
incorrect work on f412:
Numbers are milliseconds.
Here it could be seen, that after first USBRESET MCU receives GET DEV DESCR request, host receives reply then it issues a second USBRESET and  nothing happens.  After about a second I perform MCU reset and host issues USBRESET and once again USBRESET and send SETUP ADDRESS. 
It is very strange. Seems like host waits for some line state. Or maybe MCU does not see SETUP ADDRESS from first attempt.
--------------------------------------------------|6|SRQ/CIDSCHG/PTXFE/EOPF/USBSUSP/ESUSP/NPTXFE/
-|121|SRQ/CIDSCHG/PTXFE/RSTDET/EOPF/ENUMDNE/USBRST/ESUSP/NPTXFE//ED//R/
---|485|PTXFE/EOPF/NPTXFE/RXFLVL/SOF/
|485|PTXFE/EOPF/NPTXFE/RXFLVL/
|486|PTXFE/IEP/EOPF/NPTXFE/SOF/[80/06/00/40/100/]

|487|PTXFE/EOPF/NPTXFE/RXFLVL/
|487|PTXFE/EOPF/USBRST/NPTXFE/RXFLVL//R/
|496|PTXFE/ENUMDNE/NPTXFE//ED/
-|564|PTXFE/EOPF/NPTXFE/RXFLVL/SOF/
----------|6|SRQ/CIDSCHG/PTXFE/EOPF/USBSUSP/ESUSP/NPTXFE/
----|441|SRQ/CIDSCHG/PTXFE/RSTDET/EOPF/ENUMDNE/USBRST/ESUSP/NPTXFE//ED//R/
|461|PTXFE/EOPF/ENUMDNE/USBRST/NPTXFE/SOF//ED//R/
---|718|PTXFE/EOPF/NPTXFE/RXFLVL/SOF/
|718|PTXFE/EOPF/NPTXFE/RXFLVL/
|718|PTXFE/IEP/EOPF/NPTXFE/[00/05/00/00/02/]
ADDR 2

|736|PTXFE/EOPF/NPTXFE/RXFLVL/SOF/