libopencm3/libopencm3-examples #223: usb_cdcacm example on STM32F303RD not working

arendwierks, 2021-01-26:
If I upload the code to my STM32F303RDT6 board the USB is not working. Windows returns an error, and the usb device is not recognized in device manager.
When I create a project via STM32Cube IDE the usb cdc works fine, so I believe something is wrong/missing in this driver.

manuelbl, 2021-01-26:
The STM32 F3 series uses two different memory organizations for USB buffers. For this series, *libopencm3* only supports one of them, and it's not the one used in your MCU. So I'm afraid the USB stack does not work on your board. 

Also see this comment: 
https://github.com/libopencm3/libopencm3/blob/master/include/libopencm3/stm32/common/st_usbfs_v1.h#L26

The mentioned MCUs use an implementation that is more or less the *st_usbfs_v2* implementation. However, the current *libopencm3* always uses the *st_usbfs_v1* implementation for the F3 series.

If you want to modify the library, have a look at issue https://github.com/libopencm3/libopencm3/issues/841.

arendwierks, 2021-01-27:
Hi Manuel,
Thanks for the help, I didn't find Gussy's issue yesterday, but I will try that.