libopencm3/libopencm3-examples #234: arm-none-eabi-gcc not found error

shane-woods, 2022-05-27:
Hello,

I️ am just getting started out using libopencm3 and I️ have run into an issue when trying to run the Makefile at the top level of **libopencm3_examples**. After running git subdirectory init and git subdirectory update to have the /libopenmc3 files as a subdirectory, I call make. However when I️ do so, I️ am met with this error multiple times:

**make[2]: arm-none-eabi-gcc: No such file or directory**

The odd part is that I️ was running into this issue when running make at the top level of the **libopencm3** directory. And I️ soon after realized it was because I️ had installed the ARM toolchain on my M1 Macbook and I️ guess the toolchain only works on Intel Macs so far. 

I️ ran a script I️ found on github to get the xPack distribution of the ARM toolchain which runs with Apple M1's. I️ then ran make again on the **libopencm3** directory and it successfully compiled. So why now does it not work for **libopencm3_examples**? 

If anyone has any ideas as to why this might be happening, please let me know.

Thanks,
Shane

todbot, 2022-05-27:
Seems like you don't actually have the gcc-arm toolchain installed. I'd recommend using [Homebrew](https://brew.sh/) if you're not already using it and doing:  `brew install gcc-arm-embedded`.

Alternatively, if you already have Arduino installed, you can use the GCC ARM that's there, with something like:
```sh
export ARM_GCC_HOME=${HOME}/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4
export PATH=${PATH}:${ARM_GCC_HOME}/bin
```
(the actual path may depend on which exact version of Arduino you have installed)

shane-woods, 2022-05-27:
So I️ was able to run make and it compiled. I️ think the changes I️ had made had not synced with my text editor as I️ was bouncing between vscode's terminal and the terminal on mac.

I️ installed the gcc-arm-embedded through the ARM website, I️ can try again on brew. The part I️ am confused about from @karlp 's response is that the ARM website said it only supports x86 Macs. So wouldn't it matter what hardware I️ was on? Sorry I️'m a bit new to this.

And @todbot I️ am using vscode for stm32 so I️ don't think that part about Arduino applies to me

todbot, 2022-05-27:
I mean that if you already have Arduino installed, it ships with an ARM GCC compiler that works on STM32, EFM32, SAMD, etc. chips. The compiler is universal in that regard.

And yes, the GCC ARM compilers I'm aware of are compiled for x64 but work just fine on an M1 Mac. (I'm using that compiler right now on my M1 MBP)

shane-woods, 2022-05-27:
Okay, thanks for the quick response I'm going to try again with the x64 download through homebrew

todbot, 2022-05-27:
If you already have a download from ARM, that will work. Just maybe sounds like your PATH wasn't modified to see it.

shane-woods, 2022-05-27:
Okay thank you

closed 2022-05-27 by shane-woods