libopencm3/libopencm3-examples #235: Unable to run elf files

shane-woods, 2022-05-31:
Hello,

I️ have read through the /libopencm3 wiki and have not found an solution, but I️ am working with a stm32f0 and trying to run some of the example programs in this repo, but when I️ compile and try to execute the .elf file, I️ run into this error.

./button.elf: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), statically linked, with debug_info, not stripped.

I️ am working with a 64 bit machine so I️ am wondering if anyone knows any solutions to this?

shane-woods, 2022-05-31:
Oh okay, I️ had flashed the pin configuration for our board via the STM32CubeIDE but I️ can try that. Do you know if there is any instructions in the wiki/docs to do what you are saying?

todbot, 2022-05-31:
https://github.com/libopencm3/libopencm3-examples#make-flash-target

closed 2022-05-31 by shane-woods