libopencm3/libopencm3-examples #242: Composite USB device example

PocketPi, 2023-03-07:
Hey

I am trying to make a composite usb device consisting of cdcacm and msc profiles.

I am using a STM32F411CEU 

I have tried to merge the cdcacm and msc example but when i connect my device to the pc i get the following error: 

```
Mar 07 13:47:02 thickpad kernel: usb 3-6.1.4: new full-speed USB device number 36 using xhci_hcd
Mar 07 13:47:02 thickpad kernel: usb 3-6.1.4: New USB device found, idVendor=aabb, idProduct=ccdd, bcdDevice= 2.00
Mar 07 13:47:02 thickpad kernel: usb 3-6.1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Mar 07 13:47:02 thickpad kernel: usb 3-6.1.4: Product: wqer
Mar 07 13:47:02 thickpad kernel: usb 3-6.1.4: Manufacturer: asdf
Mar 07 13:47:02 thickpad kernel: usb 3-6.1.4: SerialNumber: 62908886
Mar 07 13:47:02 thickpad kernel: usb-storage 3-6.1.4:1.1: USB Mass Storage device detected
Mar 07 13:47:02 thickpad kernel: scsi host0: usb-storage 3-6.1.4:1.1
Mar 07 13:47:34 thickpad kernel: usb 3-6.1.4: reset full-speed USB device number 36 using xhci_hcd
Mar 07 13:47:49 thickpad kernel: usb 3-6.1.4: device descriptor read/64, error -110
Mar 07 13:48:05 thickpad kernel: usb 3-6.1.4: device descriptor read/64, error -110
Mar 07 13:48:05 thickpad kernel: usb 3-6.1.4: reset full-speed USB device number 36 using xhci_hcd
```
This is most of the code i am running:
```
#include "usb.h"
#include "buffered_printf.h"
#include "ramdisk.h"
#include <libopencm3/cm3/nvic.h>
#include <libopencm3/usb/cdc.h>
// clang-format off
#include <libopencm3/usb/usbd.h>
#include <libopencm3/usb/msc.h>
// clang-format on
#include <stddef.h>

#define DEVICE_ID_ADDR 0x1FFF7A10

static char serialno[9] = {0};

static usbd_device *usb_device;

static const struct usb_device_descriptor dev_descr = {
    .bLength = USB_DT_DEVICE_SIZE,
    .bDescriptorType = USB_DT_DEVICE,
    .bcdUSB = 0x0200,
    .bDeviceClass = 0,
    .bDeviceSubClass = 0,
    .bDeviceProtocol = 0,
    .bMaxPacketSize0 = 64,
    .idVendor = 0xaabb,
    .idProduct = 0xccdd,
    .bcdDevice = 0x0200,
    .iManufacturer = 1,
    .iProduct = 2,
    .iSerialNumber = 3,
    .bNumConfigurations = 1,
};

static const struct usb_endpoint_descriptor data_endp[] = {{
                                                               .bLength = USB_DT_ENDPOINT_SIZE,
                                                               .bDescriptorType = USB_DT_ENDPOINT,
                                                               .bEndpointAddress = 0x01,
                                                               .bmAttributes = USB_ENDPOINT_ATTR_BULK,
                                                               .wMaxPacketSize = 64,
                                                               .bInterval = 1,
                                                           },
                                                           {
                                                               .bLength = USB_DT_ENDPOINT_SIZE,
                                                               .bDescriptorType = USB_DT_ENDPOINT,
                                                               .bEndpointAddress = 0x81,
                                                               .bmAttributes = USB_ENDPOINT_ATTR_BULK,
                                                               .wMaxPacketSize = 64,
                                                               .bInterval = 1,
                                                           }};

static const struct usb_endpoint_descriptor msc_endp[] = {{
                                                              .bLength = USB_DT_ENDPOINT_SIZE,
                                                              .bDescriptorType = USB_DT_ENDPOINT,
                                                              .bEndpointAddress = 0x03,
                                                              .bmAttributes = USB_ENDPOINT_ATTR_BULK,
                                                              .wMaxPacketSize = 64,
                                                              .bInterval = 1,
                                                          },
                                                          {
                                                              .bLength = USB_DT_ENDPOINT_SIZE,
                                                              .bDescriptorType = USB_DT_ENDPOINT,
                                                              .bEndpointAddress = 0x83,
                                                              .bmAttributes = USB_ENDPOINT_ATTR_BULK,
                                                              .wMaxPacketSize = 64,
                                                              .bInterval = 1,
                                                          }};

static const struct usb_interface_descriptor data_iface[] = {{
    .bLength = USB_DT_INTERFACE_SIZE,
    .bDescriptorType = USB_DT_INTERFACE,
    .bInterfaceNumber = 0,
    .bAlternateSetting = 0,
    .bNumEndpoints = 2,
    .bInterfaceClass = USB_CLASS_DATA,
    .bInterfaceSubClass = 0,
    .bInterfaceProtocol = 0,
    .iInterface = 0,

    .endpoint = data_endp,
}};

static const struct usb_interface_descriptor msc_iface[] = {{
    .bLength = USB_DT_INTERFACE_SIZE,
    .bDescriptorType = USB_DT_INTERFACE,
    .bInterfaceNumber = 1,
    .bAlternateSetting = 0,
    .bNumEndpoints = 2,
    .bInterfaceClass = USB_CLASS_MSC,
    .bInterfaceSubClass = USB_MSC_SUBCLASS_SCSI,
    .bInterfaceProtocol = USB_MSC_PROTOCOL_BBB,
    .iInterface = 0,

    .endpoint = msc_endp,
}};

static const struct usb_interface ifaces[] = {{
                                                  .num_altsetting = 1,
                                                  .altsetting = data_iface,
                                              },
                                              {
                                                  .num_altsetting = 1,
                                                  .altsetting = msc_iface,
                                              }};

static const struct usb_config_descriptor config_descr = {
    .bLength = USB_DT_CONFIGURATION_SIZE,
    .bDescriptorType = USB_DT_CONFIGURATION,
    .wTotalLength = 0,
    .bNumInterfaces = 2,
    .bConfigurationValue = 1,
    .iConfiguration = 0,
    .bmAttributes = 0x80,
    .bMaxPower = 0x32,

    .interface = ifaces,
};

static const char *usb_strings[] = {
    "asdf",
    "wqer",
    serialno,
};

/* Buffer to be used for control requests. */
uint8_t usbd_control_buffer[128];

static void cdcacm_sof_callback(void) {
    char buf[64];
    uint16_t i = 0;
    while (!usb_fifo_is_empty() && i < 64) {
        usb_fifo_get(&buf[i++]);
    }

    if (i && usb_device) {
        while (usbd_ep_write_packet(usb_device, 0x82, buf, i) == 0) {
        };
    }
}

static void cdcacm_data_rx_cb(usbd_device *usbd_dev, uint8_t ep) {
    (void)ep;

    char buf[64];
    uint16_t len = usbd_ep_read_packet(usbd_dev, 0x01, buf, 64);
}

static void cdcacm_set_config(usbd_device *usbd_dev, uint16_t wValue) {
    (void)wValue;

    usbd_ep_setup(usbd_dev, 0x01, USB_ENDPOINT_ATTR_BULK, 64, cdcacm_data_rx_cb);
    usbd_ep_setup(usbd_dev, 0x81, USB_ENDPOINT_ATTR_BULK, 64, NULL);

    usbd_register_sof_callback(usbd_dev, cdcacm_sof_callback);
}

static void serialno_read(char *buf) {
    volatile uint32_t *DEVICE_ID = (volatile uint32_t *)DEVICE_ID_ADDR;
    const uint32_t uid = *DEVICE_ID + *(DEVICE_ID + 1) + *(DEVICE_ID + 2);

    /* Fetch serial number from chip's unique ID. */
    for (int i = 0; i < 8; i++)
        buf[7 - i] = ((uid >> (4 * i)) & 0xF) + '0';

    for (int i = 0; i < 8; i++)
        if (buf[i] > '9')
            buf[i] += 'A' - '9' - 1;
    buf[8] = 0;
}

void usb_setup(void) {
    serialno_read(serialno);

    usb_device = usbd_init(
        &otgfs_usb_driver, &dev_descr, &config_descr, usb_strings, 3, usbd_control_buffer, sizeof(usbd_control_buffer));

    usbd_register_set_config_callback(usb_device, cdcacm_set_config);

    ramdisk_init();
    usb_msc_init(
        usb_device, 0x83, 64, 0x03, 64, "VendorID", "ProductID", "0.00", ramdisk_blocks(), ramdisk_read, ramdisk_write);

    /* NVIC setup. */
    nvic_enable_irq(NVIC_OTG_FS_IRQ);
    nvic_set_priority(NVIC_OTG_FS_IRQ, 1);
}

void otg_fs_isr(void) {
    if (usb_device) {
        usbd_poll(usb_device);
    }
}
```

Can you guys help me find out how to progress either with concrete changes to the code or how to debug the problem.

karlp, 2023-03-07:
You probably need to make your control buffer bigger, as your descriptors will now exceed it's space.   until https://github.com/libopencm3/libopencm3/pull/1140 is merged, you need to have space to build the descriptors in ram for talking to the host

PocketPi, 2023-03-07:
changing the control buffer size to 2048 does not solve the problem.

karlp, 2023-03-07:
I would recheck your descriptors carefully then, I don't have any other ideas, and I've not personally setup a composite with msc. you might want to check that usb_msc_init isn't making any assumptions internally? it's the only class handler that offers such an init() method.

PocketPi, 2023-03-07:
CDC and MIDI works together. 

I shall try to figure out why MSC does not work

Tubatstuff, 2023-11-19:
Did you ever resolve this one?  I'm looking or a CDCACM+MSC composite solution using libopencm3 currently.

PocketPi, 2023-11-22:
> Did you ever resolve this one? I'm looking or a CDCACM+MSC composite solution using libopencm3 currently.

yes i did. 

Its a while ago, but i think that the main problem was not assigning the correct values to the structs. 

Here is a copy of what i got working 

```
#define SIZEOFARRAY(x) (sizeof(x) / sizeof((x)[0]))

#define USB_DATA_ENDPOINT_IN  0x81
#define USB_DATA_ENDPOINT_OUT 0x01
#define USB_MSC_ENDPOINT_IN   0x82
#define USB_MSC_ENDPOINT_OUT  0x02
#define USB_CDC_ENDPOINT_IN   0x83

#define INTERFACE_NO_1 0
#define INTERFACE_NO_2 2

#define USB_PACKET_SIZE 64

static usb_endpoint_data_callback_t usb_data_callback = NULL;

static char usb_serial_number[13]; /* 12 bytes of desig and a \0 */

static usbd_device *usb_device;

static bool usb_is_connected = false;
static const struct usb_device_descriptor dev_descr = {
    .bLength = USB_DT_DEVICE_SIZE,
    .bDescriptorType = USB_DT_DEVICE,
    .bcdUSB = 0x0200,
    .bDeviceClass = USB_CLASS_CDC,
    .bDeviceSubClass = 0,
    .bDeviceProtocol = 0,
    .bMaxPacketSize0 = 64,
    .idVendor = 0xaabb,
    .idProduct = 0xccdd,
    .bcdDevice = 0x0200,
    .iManufacturer = 1,
    .iProduct = 2,
    .iSerialNumber = 3,
    .bNumConfigurations = 1,
};

static const struct {
    struct usb_cdc_header_descriptor header;
    struct usb_cdc_call_management_descriptor call_mgmt;
    struct usb_cdc_acm_descriptor acm;
    struct usb_cdc_union_descriptor cdc_union;
} __attribute__((packed))
cdcacm_functional_descriptors = {.header =
                                     {
                                         .bFunctionLength = sizeof(struct usb_cdc_header_descriptor),
                                         .bDescriptorType = CS_INTERFACE,
                                         .bDescriptorSubtype = USB_CDC_TYPE_HEADER,
                                         .bcdCDC = 0x0110,
                                     },
                                 .call_mgmt =
                                     {
                                         .bFunctionLength = sizeof(struct usb_cdc_call_management_descriptor),
                                         .bDescriptorType = CS_INTERFACE,
                                         .bDescriptorSubtype = USB_CDC_TYPE_CALL_MANAGEMENT,
                                         .bmCapabilities = 0,
                                         .bDataInterface = INTERFACE_NO_1 + 1,
                                     },
                                 .acm =
                                     {
                                         .bFunctionLength = sizeof(struct usb_cdc_acm_descriptor),
[... 254 lines cut ...]
        usbd_disconnect(usb_device, true);
    }
}

usb_error_t usb_send_serial(char *data, uint16_t len) {
    uint16_t i = 0;
    // Check if we have a valid connection
    if (!usb_device || !usb_is_connected) {
        return USB_NOT_CONNECTED;
    }

    // try to send the data @TODO no timeout.
    while (*data && (i < len)) {
        uint32_t start_time = time_now_ms();
        while (usbd_ep_write_packet(usb_device, USB_DATA_ENDPOINT_IN, (uint8_t *)data, 1) == 0) {
            if (time_elapsed_ms_check(start_time, 100)) {
                return USB_TIMEOUT;
            }
        };
        i++;
    }
    return USB_SUCCESS;
}

```