libopencm3/libopencm3-examples #246: How stm32F429IGT6 use st-link and openocd  to burn miniblink.elf ?

tonychen0924, 2023-09-21:
Why I only generate miniblink.elf wihtout .miniblink.hex ? How to burn into stm32F429IGT6 ?

===openocd terminal===
$ pwd
/usr/share/openocd/scripts
$ openocd -f interface/stlink.cfg -f target/stm32f4x.cfg 
Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 2000 kHz
Info : STLINK V2J37S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.237209
Info : stm32f4x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f4x.cpu on 3333
Info : Listening on port 3333 for gdb connections
^Cshutdown command invoked

===make terminal
tonychen@tonychen-ASUS-TUF-Gaming-F15-FX507VV-FX507VV:~/gitee/libopencm3-examples/examples/stm32/f4/stm32f429i-discovery/miniblink$ make
tonychen@tonychen-ASUS-TUF-Gaming-F15-FX507VV-FX507VV:~/gitee/libopencm3-examples/examples/stm32/f4/stm32f429i-discovery/miniblink$ ll -rlt
总计 252
-rw-rw-r--  1 tonychen tonychen    215  9月 20 13:43 README.md
-rw-rw-r--  1 tonychen tonychen   2265  9月 20 13:43 miniblink.c
-rw-rw-r--  1 tonychen tonychen    874  9月 20 13:43 Makefile
drwxrwxr-x 22 tonychen tonychen   4096  9月 20 13:43 ../
-rw-rw-r--  1 tonychen tonychen   2208  9月 21 11:10 miniblink.d
-rw-rw-r--  1 tonychen tonychen  75740  9月 21 11:10 miniblink.o
drwxrwxr-x  2 tonychen tonychen   4096  9月 21 11:10 ./
-rw-rw-r--  1 tonychen tonychen  56312  9月 21 11:10 miniblink.map
-rwxrwxr-x  1 tonychen tonychen 156224  9月 21 11:10 miniblink.elf*
tonychen@tonychen-ASUS-TUF-Gaming-F15-FX507VV-FX507VV:~/gitee/libopencm3-examples/examples/stm32/f4/stm32f429i-discovery/miniblink$ pwd
/home/tonychen/gitee/libopencm3-examples/examples/stm32/f4/stm32f429i-discovery/miniblink
tonychen@tonychen-ASUS-TUF-Gaming-F15-FX507VV-FX507VV:~/gitee/libopencm3-examples/examples/stm32/f4/stm32f429i-discovery/miniblink$ telnet localhost 4444
Trying 127.0.0.1...
Connected to localhost.
Escape character is '^]'.
Open On-Chip Debugger
> resset halt
invalid command name "resset"
> flash write_image erase miniblink.elf
couldn't open miniblink.elf
auto erase enabled
> program miniblink.elf
Unable to match requested speed 2000 kHz, using 1800 kHz
Unable to match requested speed 2000 kHz, using 1800 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08000258 msp: 0x20030000
Unable to match requested speed 8000 kHz, using 4000 kHz
Unable to match requested speed 8000 kHz, using 4000 kHz
** Programming Started **
couldn't open miniblink.elf
embedded:startup.tcl:530: Error: ** Programming Failed **
in procedure 'program' 
in procedure 'program_error' called at file "embedded:startup.tcl", line 595
at file "embedded:startup.tcl", line 530

tonychen0924, 2023-09-21:
I update absolute path to try . LED still not responding ! What mean "Unable to match requested speed 2000 kHz, using 1800 kHz" ?

tonychen@tonychen-ASUS-TUF-Gaming-F15-FX507VV-FX507VV:~/gitee/libopencm3-examples/examples/stm32/f4/stm32f429i-discovery/miniblink$ telnet localhost 4444
Trying 127.0.0.1...
Connected to localhost.
Escape character is '^]'.
Open On-Chip Debugger
&gt; reset halt
Unable to match requested speed 2000 kHz, using 1800 kHz
Unable to match requested speed 2000 kHz, using 1800 kHz
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x08000258 msp: 0x20030000
&gt; flash write_image erase /home/tonychen/gitee/libopencm3-examples/examples/stm32/f4/stm32f429i-discovery/miniblink/miniblink.elf
auto erase enabled
wrote 16384 bytes from file /home/tonychen/gitee/libopencm3-examples/examples/stm32/f4/stm32f429i-discovery/miniblink/miniblink.elf in 0.646297s (24.756 KiB/s)

&gt; reset
Unable to match requested speed 2000 kHz, using 1800 kHz
Unable to match requested speed 2000 kHz, using 1800 kHz

karlp, 2023-09-21:
use openocd's "program" command  https://openocd.org/doc/html/Flash-Programming.html

a .elf has all the information needed for both programming and debugging.  A .hex is only useful for programming, and is simply an unnecessary step.

closed 2023-09-21 by karlp

tonychen0924, 2023-09-22:
This is document usage::
# program and verify using elf/hex/s19. verify and reset
# are optional parameters
openocd -f board/stm32f3discovery.cfg \
	-c "program filename.elf verify reset exit"
What is "board/stm32f3discovery.cfg" ??? I try with "/usr/share/openocd/scripts/target/stm32f4x.cfg" .

tonychen@tonychen-ASUS-TUF-Gaming-F15-FX507VV-FX507VV:~/gitee/libopencm3-examples/examples/stm32/f4/stm32f429i-discovery/miniblink$ openocd -f /usr/share/openocd/scripts/target/stm32f4x.cfg -c "program /home/tonychen/gitee/libopencm3-examples/examples/stm32/f4/stm32f429i-discovery/miniblink/miniblink.elf"
Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Error: Debug adapter does not support any transports? Check config file order.
Error: unable to select a session transport. Can't continue.
shutdown command invoked

tonychen@tonychen-ASUS-TUF-Gaming-F15-FX507VV-FX507VV:~/gitee/libopencm3-examples/examples/stm32/f4/stm32f429i-discovery/miniblink$ openocd -f /usr/share/openocd/scripts/interface/stlink.cfg -f /usr/share/openocd/scripts/target/stm32f4x.cfg -c "program /home/tonychen/gitee/libopencm3-examples/examples/stm32/f4/stm32f429i-discovery/miniblink/miniblink.elf"
Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select &lt;transport&gt;'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 2000 kHz
Info : STLINK V2J37S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.238804
Info : stm32f4x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f4x.cpu on 3333
Info : Listening on port 3333 for gdb connections
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08000258 msp: 0x20030000
Info : Unable to match requested speed 8000 kHz, using 4000 kHz
Info : Unable to match requested speed 8000 kHz, using 4000 kHz
** Programming Started **
Info : device id = 0x20016419
Info : flash size = 1024 kbytes
Info : Single Bank 1024 kiB STM32F42x/43x/469/479 found
** Programming Finished **
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
^Cshutdown command invoked

tonychen0924, 2023-09-22:
I update this new command to run . After LED also can not blink .  Is it caused by "Info : Unable to match requested speed 8000 kHz, using 4000 kHz"? My board is Apollo stm32F429IGT6 . And I used Keil MDK to test the normal burning program before.

$ openocd -f /usr/share/openocd/scripts/board/stm32f429discovery.cfg -c "program /home/tonychen/gitee/libopencm3-examples/examples/stm32/f4/stm32f429i-discovery/miniblink/miniblink.elf verify reset exit"
Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
srst_only separate srst_nogate srst_open_drain connect_deassert_srst

Info : clock speed 2000 kHz
Info : STLINK V2J37S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.237209
Info : stm32f4x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f4x.cpu on 3333
Info : Listening on port 3333 for gdb connections
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08000258 msp: 0x20030000
Info : Unable to match requested speed 8000 kHz, using 4000 kHz
Info : Unable to match requested speed 8000 kHz, using 4000 kHz
** Programming Started **
Info : device id = 0x20016419
Info : flash size = 1024 kbytes
Info : Single Bank 1024 kiB STM32F42x/43x/469/479 found
** Programming Finished **
** Verify Started **
** Verified OK **
** Resetting Target **
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
shutdown command invoked

tonychen0924, 2023-09-25:
@karlp

karlp, 2023-09-25:
is your apollo board an f429 discovery? does it have leds on the same pins as the f429 discovery board?  Check that.  The speed mismatch is an informational, it is not a problem.

tonychen0924, 2023-09-27:
Year, I work in "libopencm3-examples/examples/stm32/f4/stm32f429i-discovery/miniblink" . It rely “libopencm3-examples/libopencm3/include/libopencm3/stm32/f4/rcc.h” and "libopencm3-examples/libopencm3/include/libopencm3/stm32/common/gpio_common_f24.h" ？

Minilink.c use "GPIOG" , "GPIO13" . But my board supply such as "PA0","PC7" . How to map ?

karlp, 2023-09-27:
edit the file....

tonychen0924, 2023-09-28:
I update part of code in miniblink,c . Now ti can make and run successfully .

#include &lt;libopencm3/stm32/rcc.h&gt;
#include &lt;libopencm3/stm32/gpio.h&gt;

static void gpio_setup(void)
{
        /* Enable GPIOB clock. */
        rcc_periph_clock_enable(RCC_GPIOB);

        /* Set GPIO0 (in GPIO port B) to 'output push-pull'. */
        gpio_mode_setup(GPIOB, GPIO_MODE_OUTPUT, GPIO_PUPD_NONE, GPIO0);
}

int main(void)
{
        int i;

        gpio_setup();

        /* Blink the LED (PB0) on the board. */
        while (1)
        {
                /* Toggle the LED on/off using libopencm3 functions. */
                gpio_toggle(GPIOB, GPIO0); /* LED on/off */

                /* Delay to create the blinking effect. */
                for (i = 0; i &lt; 1000000; i++)
                {
                        __asm__("nop");
                }
        }

        return 0;
}